j-kaltes/Juggluco #4: root permission

q601180252, 2023-02-08:
On the root permission phone, you must disable the root permission to use libcalibrat2.so

j-kaltes, 2023-02-08:
No, on a new arm64 phone you don't need to hide root, but you can better do so. On an old arm32 phone I had to hide root, probably because it uses an old Linux version.

q601180252, 2023-02-08:
/A/libc: Fatal signal 4 (SIGILL), code 1 (ILL_ILLOPC), fault addr 0x7929fb7bc8 in tid 12530 (), pid 12530 ()
 A/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
A/DEBUG: Build fingerprint: 'OnePlus/OnePlus8T/OnePlus8T:13/RKQ1.211119.001/R.c9b6f1_9-1db5a:user/release-keys'
 A/DEBUG: Revision: '0'
 A/DEBUG: ABI: 'arm64'
A/DEBUG: Timestamp: 2023-02-08 18:41:09.764757746+0800
A/DEBUG: Process uptime: 2s
 A/DEBUG: Cmdline: 
 A/DEBUG: pid: 12530, tid: 12530, name:   >>>  <<<
A/DEBUG: uid: 10500
A/DEBUG: signal 4 (SIGILL), code 1 (ILL_ILLOPC), fault addr 0x0000007929fb7bc8
A/DEBUG:     x0  0000000000000993  x1  0000000098bcbf91  x2  000000009a0d6f5f  x3  00000000e51a7375
A/DEBUG:     x4  00000000a0fef17b  x5  00000000fb44a646  x6  000000792a08fc80  x7  000000000000000a
A/DEBUG:     x8  0000007929fb606c  x9  0000000000000006  x10 0000000000000309  x11 000000000000004a
A/DEBUG:     x12 0000007fd3804d6b  x13 000000000000005a  x14 00000000ff17d796  x15 000000005610f663
A/DEBUG:     x16 0000000000007865  x17 0000007929fbb074  x18 0000007c6ea4e000  x19 0000007929fb7bc8
A/DEBUG:     x20 0000007929fb7bc8  x21 00000000fb063959  x22 0000000000000059  x23 0000007fd3804d6b
A/DEBUG:     x24 00000000e6029bd2  x25 0000000019fd6421  x26 000000792bdc8c00  x27 00000000b59c4ce8
A/DEBUG:     x28 00000000e6029bdf  x29 0000007929fb7bc8
A/DEBUG:     lr  0000007929fb7bc8  sp  0000007fd3804b90  pc  0000007929fb7bc8  pst 0000000080001000
A/DEBUG: backtrace:
A/DEBUG:       #00 pc 0000000000001bc8  <anonymous:7929fb6000>

q601180252, 2023-02-08:
[image]

j-kaltes, 2023-02-08:
What when you try again?

q601180252, 2023-02-08:
Yes, I used the US libre 2 sensor
> What when you try a second time? How do you use it, because you use a US libre 2 sensor or because you have made it do so yourself in some way?

j-kaltes, 2023-02-08:
On my phone it doesn't crash with root. Did you already use Juggluco already before its source was available? How long ago did it crash?

q601180252, 2023-02-08:
I have not used the US libre 2 sensor with Juggluco  ,I'll try it with Juggluco tomorrow

j-kaltes, 2023-02-08:
How then did you get this logcat?

q601180252, 2023-02-08:
I modified 'jnisub.cpp'
struct packagepath {
    static constexpr const char packstart[] = "package:";
//    static constexpr const char lib[] = "/libcalibrate.so";
    static constexpr const char lib[] = "/libcalibrat2.so";

byte[] uid = hexStringToBytes("52EEAE0200A407E0");
        byte[] info = hexStringToBytes("9D083001E61F");
        byte[] data =hexStringToBytes("552dcf2c89408798797bd44f7920b2d25202cace3c951c4c190406ac168a78b25056c142cf882479280dbf23ab18363401232a61e96534af89137970ad3822715dd3a18c06d58e49a87cace297b20a6c14f1df64671c3c9a4865a21b7f18699717ad5de833ea0aafc7f9b5996771e9e50bb30a1cb356af64ae7e9c8a2a530e91c841996e9e41523ef61af943fac12873339c319eed735f5e80f792f1a79cade03c37490d4c9967d8e070056394fec4adf75961364f9ba015db03ade14b098edf7048129ef3f47f2ca09063e0539092a29de100858129f435796611019c3880765fbd35eae550dd096de975e8c1a4ac4b653f7a204e6f2677b34d09460e61f5c5ed82c571a49075f2d322c8dba0f8dd782eafd05dc4e5a52e8634c6f6dd5d0b2c6c0b38d132a0dd07bca772afc4f07f51009537d534b2ffd6b48e94e34ee79b138a9d5b01ad0e2b7d162626f68065a8c037c9b4f79f31f390");

Natives.nfcdata(uid, info, data)

j-kaltes, 2023-02-08:
Your code also crashes with me. The following doesn't crash immediately:

package tk.glucodata;

class test {
	static private final String LOG_ID="test";
static	void test() {
                var thr=new Thread(()-> testthread());
                thr.start();
                }

	static void testthread() {

		byte[] uid = {(byte)0x52,(byte)0xEE,(byte)0xAE,(byte)0x02,(byte)0x00,(byte)0xA4,(byte)0x07,(byte)0xE0};
		byte[] info = {(byte)0x9D,(byte)0x08,(byte)0x30,(byte)0x01,(byte)0xE6,(byte)0x1F};
		byte[] data = {(byte)0x55,(byte)0x2d,(byte)0xcf,(byte)0x2c,(byte)0x89,(byte)0x40,(byte)0x87,(byte)0x98,(byte)0x79,(byte)0x7b,(byte)0xd4,(byte)0x4f,(byte)0x79,(byte)0x20,(byte)0xb2,(byte)0xd2,(byte)0x52,(byte)0x02,(byte)0xca,(byte)0xce,(byte)0x3c,(byte)0x95,(byte)0x1c,(byte)0x4c,(byte)0x19,(byte)0x04,(byte)0x06,(byte)0xac,(byte)0x16,(byte)0x8a,(byte)0x78,(byte)0xb2,(byte)0x50,(byte)0x56,(byte)0xc1,(byte)0x42,(byte)0xcf,(byte)0x88,(byte)0x24,(byte)0x79,(byte)0x28,(byte)0x0d,(byte)0xbf,(byte)0x23,(byte)0xab,(byte)0x18,(byte)0x36,(byte)0x34,(byte)0x01,(byte)0x23,(byte)0x2a,(byte)0x61,(byte)0xe9,(byte)0x65,(byte)0x34,(byte)0xaf,(byte)0x89,(byte)0x13,(byte)0x79,(byte)0x70,(byte)0xad,(byte)0x38,(byte)0x22,(byte)0x71,(byte)0x5d,(byte)0xd3,(byte)0xa1,(byte)0x8c,(byte)0x06,(byte)0xd5,(byte)0x8e,(byte)0x49,(byte)0xa8,(byte)0x7c,(byte)0xac,(byte)0xe2,(byte)0x97,(byte)0xb2,(byte)0x0a,(byte)0x6c,(byte)0x14,(byte)0xf1,(byte)0xdf,(byte)0x64,(byte)0x67,(byte)0x1c,(byte)0x3c,(byte)0x9a,(byte)0x48,(byte)0x65,(byte)0xa2,(byte)0x1b,(byte)0x7f,(byte)0x18,(byte)0x69,(byte)0x97,(byte)0x17,(byte)0xad,(byte)0x5d,(byte)0xe8,(byte)0x33,(byte)0xea,(byte)0x0a,(byte)0xaf,(byte)0xc7,(byte)0xf9,(byte)0xb5,(byte)0x99,(byte)0x67,(byte)0x71,(byte)0xe9,(byte)0xe5,(byte)0x0b,(byte)0xb3,(byte)0x0a,(byte)0x1c,(byte)0xb3,(byte)0x56,(byte)0xaf,(byte)0x64,(byte)0xae,(byte)0x7e,(byte)0x9c,(byte)0x8a,(byte)0x2a,(byte)0x53,(byte)0x0e,(byte)0x91,(byte)0xc8,(byte)0x41,(byte)0x99,(byte)0x6e,(byte)0x9e,(byte)0x41,(byte)0x52,(byte)0x3e,(byte)0xf6,(byte)0x1a,(byte)0xf9,(byte)0x43,(byte)0xfa,(byte)0xc1,(byte)0x28,(byte)0x73,(byte)0x33,(byte)0x9c,(byte)0x31,(byte)0x9e,(byte)0xed,(byte)0x73,(byte)0x5f,(byte)0x5e,(byte)0x80,(byte)0xf7,(byte)0x92,(byte)0xf1,(byte)0xa7,(byte)0x9c,(byte)0xad,(byte)0xe0,(byte)0x3c,(byte)0x37,(byte)0x49,(byte)0x0d,(byte)0x4c,(byte)0x99,(byte)0x67,(byte)0xd8,(byte)0xe0,(byte)0x70,(byte)0x05,(byte)0x63,(byte)0x94,(byte)0xfe,(byte)0xc4,(byte)0xad,(byte)0xf7,(byte)0x59,(byte)0x61,(byte)0x36,(byte)0x4f,(byte)0x9b,(byte)0xa0,(byte)0x15,(byte)0xdb,(byte)0x03,(byte)0xad,(byte)0xe1,(byte)0x4b,(byte)0x09,(byte)0x8e,(byte)0xdf,(byte)0x70,(byte)0x48,(byte)0x12,(byte)0x9e,(byte)0xf3,(byte)0xf4,(byte)0x7f,(byte)0x2c,(byte)0xa0,(byte)0x90,(byte)0x63,(byte)0xe0,(byte)0x53,(byte)0x90,(byte)0x92,(byte)0xa2,(byte)0x9d,(byte)0xe1,(byte)0x00,(byte)0x85,(byte)0x81,(byte)0x29,(byte)0xf4,(byte)0x35,(byte)0x79,(byte)0x66,(byte)0x11,(byte)0x01,(byte)0x9c,(byte)0x38,(byte)0x80,(byte)0x76,(byte)0x5f,(byte)0xbd,(byte)0x35,(byte)0xea,(byte)0xe5,(byte)0x50,(byte)0xdd,(byte)0x09,(byte)0x6d,(byte)0xe9,(byte)0x75,(byte)0xe8,(byte)0xc1,(byte)0xa4,(byte)0xac,(byte)0x4b,(byte)0x65,(byte)0x3f,(byte)0x7a,(byte)0x20,(byte)0x4e,(byte)0x6f,(byte)0x26,(byte)0x77,(byte)0xb3,(byte)0x4d,(byte)0x09,(byte)0x46,(byte)0x0e,(byte)0x61,(byte)0xf5,(byte)0xc5,(byte)0xed,(byte)0x82,(byte)0xc5,(byte)0x71,(byte)0xa4,(byte)0x90,(byte)0x75,(byte)0xf2,(byte)0xd3,(byte)0x22,(byte)0xc8,(byte)0xdb,(byte)0xa0,(byte)0xf8,(byte)0xdd,(byte)0x78,(byte)0x2e,(byte)0xaf,(byte)0xd0,(byte)0x5d,(byte)0xc4,(byte)0xe5,(byte)0xa5,(byte)0x2e,(byte)0x86,(byte)0x34,(byte)0xc6,(byte)0xf6,(byte)0xdd,(byte)0x5d,(byte)0x0b,(byte)0x2c,(byte)0x6c,(byte)0x0b,(byte)0x38,(byte)0xd1,(byte)0x32,(byte)0xa0,(byte)0xdd,(byte)0x07,(byte)0xbc,(byte)0xa7,(byte)0x72,(byte)0xaf,(byte)0xc4,(byte)0xf0,(byte)0x7f,(byte)0x51,(byte)0x00,(byte)0x95,(byte)0x37,(byte)0xd5,(byte)0x34,(byte)0xb2,(byte)0xff,(byte)0xd6,(byte)0xb4,(byte)0x8e,(byte)0x94,(byte)0xe3,(byte)0x4e,(byte)0xe7,(byte)0x9b,(byte)0x13,(byte)0x8a,(byte)0x9d,(byte)0x5b,(byte)0x01,(byte)0xad,(byte)0x0e,(byte)0x2b,(byte)0x7d,(byte)0x16,(byte)0x26,(byte)0x26,(byte)0xf6,(byte)0x80,(byte)0x65,(byte)0xa8,(byte)0xc0,(byte)0x37,(byte)0xc9,(byte)0xb4,(byte)0xf7,(byte)0x9f,(byte)0x31,(byte)0xf3,(byte)0x90};
Log.i(LOG_ID,"before nfcdata");
		int res=Natives.nfcdata(uid, info, data);
		Log.i(LOG_ID,"after nfcdata "+res);
		Applic.app.redraw() ;
		try {
			Thread.sleep(0xFFFFFFFFFL);
		} catch (InterruptedException e) {
			Log.stack(LOG_ID,"testthread",e);
			}
		Log.i(LOG_ID,"end testthread");
	}
}

q601180252, 2023-02-09:
If I call Natives.switchgen2() directly without using nfc to connect the libre sensor, it's an infinite loop
        boolean flag = Natives.gethaslibrary();
        android.util.Log.e(LOG_ID, "flag===" + flag);
        flag = Natives.gethaslibrary();
        android.util.Log.e(LOG_ID, "flag===" + flag);
        flag = Natives.abbottinit();
        android.util.Log.e(LOG_ID, "flag===" + flag);
        flag = Natives.switchgen2();

q601180252, 2023-02-09:
> If I call Natives.switchgen2() directly without using nfc to connect the libre sensor, it's an infinite loop boolean flag = Natives.gethaslibrary(); android.util.Log.e(LOG_ID, "flag===" + flag); [...]
error log
[traceeror.log](https://github.com/j-kaltes/Juggluco/files/10694358/traceeror.log)

q601180252, 2023-02-09:
> I modified 'jnisub.cpp'
> struct packagepath {
> static constexpr const char packstart[] = "package:";
> // static constexpr const char lib[] = "/libcalibrate.so";
> static constexpr const char lib[] = "/libcalibrat2.so";

> Your code also crashes with me. The following doesn't crash immediately:
> 
> ```
> package tk.glucodata;
> 
> class test {
> 	static private final String LOG_ID="test";
>       static	void test() {
>                 var thr=new Thread(()-> testthread());
>                 thr.start();
>                 }
> 
> 	static void testthread() {
> 		byte[] uid = {(byte)0x52,(byte)0xEE,(byte)0xAE,(byte)0x02,(byte)0x00,(byte)0xA4,(byte)0x07,(byte)0xE0};
> 		byte[] info = {(byte)0x9D,(byte)0x08,(byte)0x30,(byte)0x01,(byte)0xE6,(byte)0x1F};
> 		byte[] data = {(byte)0x55,(byte)0x2d,(byte)0xcf,(byte)0x2c,(byte)0x89,(byte)0x40,(byte)0x87,(byte)0x98,(byte)0x79,(byte)0x7b,(byte)0xd4,(byte)0x4f,(byte)0x79,(byte)0x20,(byte)0xb2,(byte)0xd2,(byte)0x52,(byte)0x02,(byte)0xca,(byte)0xce,(byte)0x3c,(byte)0x95,(byte)0x1c,(byte)0x4c,(byte)0x19,(byte)0x04,(byte)0x06,(byte)0xac,(byte)0x16,(byte)0x8a,(byte)0x78,(byte)0xb2,(byte)0x50,(byte)0x56,(byte)0xc1,(byte)0x42,(byte)0xcf,(byte)0x88,(byte)0x24,(byte)0x79,(byte)0x28,(byte)0x0d,(byte)0xbf,(byte)0x23,(byte)0xab,(byte)0x18,(byte)0x36,(byte)0x34,(byte)0x01,(byte)0x23,(byte)0x2a,(byte)0x61,(byte)0xe9,(byte)0x65,(byte)0x34,(byte)0xaf,(byte)0x89,(byte)0x13,(byte)0x79,(byte)0x70,(byte)0xad,(byte)0x38,(byte)0x22,(byte)0x71,(byte)0x5d,(byte)0xd3,(byte)0xa1,(byte)0x8c,(byte)0x06,(byte)0xd5,(byte)0x8e,(byte)0x49,(byte)0xa8,(byte)0x7c,(byte)0xac,(byte)0xe2,(byte)0x97,(byte)0xb2,(byte)0x0a,(byte)0x6c,(byte)0x14,(byte)0xf1,(byte)0xdf,(byte)0x64,(byte)0x67,(byte)0x1c,(byte)0x3c,(byte)0x9a,(byte)0x48,(byte)0x65,(byte)0xa2,(byte)0x1b,(byte)0x7f,(byte)0x18,(byte)0x69,(byte)0x97,(byte)0x17,(byte)0xad,(byte)0x5d,(byte)0xe8,(byte)0x33,(byte)0xea,(byte)0x0a,(byte)0xaf,(byte)0xc7,(byte)0xf9,(byte)0xb5,(byte)0x99,(byte)0x67,(byte)0x71,(byte)0xe9,(byte)0xe5,(byte)0x0b,(byte)0xb3,(byte)0x0a,(byte)0x1c,(byte)0xb3,(byte)0x56,(byte)0xaf,(byte)0x64,(byte)0xae,(byte)0x7e,(byte)0x9c,(byte)0x8a,(byte)0x2a,(byte)0x53,(byte)0x0e,(byte)0x91,(byte)0xc8,(byte)0x41,(byte)0x99,(byte)0x6e,(byte)0x9e,(byte)0x41,(byte)0x52,(byte)0x3e,(byte)0xf6,(byte)0x1a,(byte)0xf9,(byte)0x43,(byte)0xfa,(byte)0xc1,(byte)0x28,(byte)0x73,(byte)0x33,(byte)0x9c,(byte)0x31,(byte)0x9e,(byte)0xed,(byte)0x73,(byte)0x5f,(byte)0x5e,(byte)0x80,(byte)0xf7,(byte)0x92,(byte)0xf1,(byte)0xa7,(byte)0x9c,(byte)0xad,(byte)0xe0,(byte)0x3c,(byte)0x37,(byte)0x49,(byte)0x0d,(byte)0x4c,(byte)0x99,(byte)0x67,(byte)0xd8,(byte)0xe0,(byte)0x70,(byte)0x05,(byte)0x63,(byte)0x94,(byte)0xfe,(byte)0xc4,(byte)0xad,(byte)0xf7,(byte)0x59,(byte)0x61,(byte)0x36,(byte)0x4f,(byte)0x9b,(byte)0xa0,(byte)0x15,(byte)0xdb,(byte)0x03,(byte)0xad,(byte)0xe1,(byte)0x4b,(byte)0x09,(byte)0x8e,(byte)0xdf,(byte)0x70,(byte)0x48,(byte)0x12,(byte)0x9e,(byte)0xf3,(byte)0xf4,(byte)0x7f,(byte)0x2c,(byte)0xa0,(byte)0x90,(byte)0x63,(byte)0xe0,(byte)0x53,(byte)0x90,(byte)0x92,(byte)0xa2,(byte)0x9d,(byte)0xe1,(byte)0x00,(byte)0x85,(byte)0x81,(byte)0x29,(byte)0xf4,(byte)0x35,(byte)0x79,(byte)0x66,(byte)0x11,(byte)0x01,(byte)0x9c,(byte)0x38,(byte)0x80,(byte)0x76,(byte)0x5f,(byte)0xbd,(byte)0x35,(byte)0xea,(byte)0xe5,(byte)0x50,(byte)0xdd,(byte)0x09,(byte)0x6d,(byte)0xe9,(byte)0x75,(byte)0xe8,(byte)0xc1,(byte)0xa4,(byte)0xac,(byte)0x4b,(byte)0x65,(byte)0x3f,(byte)0x7a,(byte)0x20,(byte)0x4e,(byte)0x6f,(byte)0x26,(byte)0x77,(byte)0xb3,(byte)0x4d,(byte)0x09,(byte)0x46,(byte)0x0e,(byte)0x61,(byte)0xf5,(byte)0xc5,(byte)0xed,(byte)0x82,(byte)0xc5,(byte)0x71,(byte)0xa4,(byte)0x90,(byte)0x75,(byte)0xf2,(byte)0xd3,(byte)0x22,(byte)0xc8,(byte)0xdb,(byte)0xa0,(byte)0xf8,(byte)0xdd,(byte)0x78,(byte)0x2e,(byte)0xaf,(byte)0xd0,(byte)0x5d,(byte)0xc4,(byte)0xe5,(byte)0xa5,(byte)0x2e,(byte)0x86,(byte)0x34,(byte)0xc6,(byte)0xf6,(byte)0xdd,(byte)0x5d,(byte)0x0b,(byte)0x2c,(byte)0x6c,(byte)0x0b,(byte)0x38,(byte)0xd1,(byte)0x32,(byte)0xa0,(byte)0xdd,(byte)0x07,(byte)0xbc,(byte)0xa7,(byte)0x72,(byte)0xaf,(byte)0xc4,(byte)0xf0,(byte)0x7f,(byte)0x51,(byte)0x00,(byte)0x95,(byte)0x37,(byte)0xd5,(byte)0x34,(byte)0xb2,(byte)0xff,(byte)0xd6,(byte)0xb4,(byte)0x8e,(byte)0x94,(byte)0xe3,(byte)0x4e,(byte)0xe7,(byte)0x9b,(byte)0x13,(byte)0x8a,(byte)0x9d,(byte)0x5b,(byte)0x01,(byte)0xad,(byte)0x0e,(byte)0x2b,(byte)0x7d,(byte)0x16,(byte)0x26,(byte)0x26,(byte)0xf6,(byte)0x80,(byte)0x65,(byte)0xa8,(byte)0xc0,(byte)0x37,(byte)0xc9,(byte)0xb4,(byte)0xf7,(byte)0x9f,(byte)0x31,(byte)0xf3,(byte)0x90};
>             Log.i(LOG_ID,"before nfcdata");
> 		int res=Natives.nfcdata(uid, info, data);
> 		Log.i(LOG_ID,"after nfcdata "+res);
> 		Applic.app.redraw() ;
> 		try {
> 			Thread.sleep(0xFFFFFFFFFL);
> 		} catch (InterruptedException e) {
> 			Log.stack(LOG_ID,"testthread",e);
> 			}
> 		Log.i(LOG_ID,"end testthread");
> 	}
> }
> ```
> 
> EDIT: Which library is used, I determined in another way than you.

Because I changed it
` 'jnisub.cpp'
struct packagepath {
static constexpr const char packstart[] = "package:";
static constexpr const char lib[] = "/libcalibrat2.so";`

q601180252, 2023-02-09:
> If I call Natives.switchgen2() directly without using nfc to connect the libre sensor, it's an infinite loop `
> 
> ```
>       boolean flag = Natives.gethaslibrary();
> 
>     android.util.Log.e(LOG_ID, "flag===" + flag);
> 
>     flag = Natives.abbottinit();
> 
>     android.util.Log.e(LOG_ID, "flag===" + flag);
> 
>     flag = Natives.switchgen2();
> 
>     android.util.Log.e(LOG_ID, "flag===" + flag);
> ```
> 
> `
> 
> [traceeror.log](https://github.com/j-kaltes/Juggluco/files/10694391/traceeror.log)

If I hide the root permission with magisk, it's OK,otherwise it must be called with an nfc connection

j-kaltes, 2023-02-09:
I added the following to Common/src/main/cpp/g.cpp :

#ifndef NDEBUG
#define TESTGEN2 1
#endif

q601180252, 2023-02-09:
still the same

j-kaltes, 2023-02-09:
Then use the app properly, rather than making ill-considered changes.

closed 2023-02-22 by q601180252

j-kaltes, 2023-09-09:
I found out that previous versions did crash with US Libre 2 sensors when Magisk wasn't hidden.