j-kaltes/Juggluco #8: Switch between 5 minutes and minutely broadcast

MasterPlexus, 2023-03-04:
I facing, since I use juggluco, a high amount of Akku consumption by AAPS. I think this is may linked to the situation that AAPS recalculate everything now every minute, because the values come every minute.

So my question if you could include may a switch in the settings to set between minutely broadcast or every 5 minutes, so the user could decide?
Anyhow thanks for all you did ;)

j-kaltes, 2023-03-04:
Reducing the Broadcast from once in a minute to once in 5 minutes, would probably reduce battery consumption. But battery consumption can also be reduced by getting rid of other apps running in the background by rebooting the phone now and then and deinstalling and force closing unnecessary apps:
https://www.droidwin.com/debloat-remove-bloatware-from-samsung-devices-via-adb/
https://forum.xda-developers.com/t/how-to-debloat-adb-the-ultimate-adb-debloating-thread-for-the-s20-u-series.4089089/

I made the xDrip broadcast once in a minute instead of once in 5 minutes, because I think that once in a minute is better. It often happens that the glucose level is steeply rising or falling in which case basing decisions on a value 5 minutes older would be totally wrong. Giving every 5 minutes the average of previous values, makes the produced value even older because then at the moment it is giving it is already the average of also older values. I didn't want Juggluco to produce the same values as xDrip. Naturally this also means that AAPS should be configured differently when making more often smaller adjustments.

MasterPlexus, 2023-03-04:
Thanks for your explanation, Wich I could understand. I have a Pixel 4 xl and in the past the Akku was enough for 1,5 day, now in a half day I have to reload.

But again I understand your point.

closed 2023-03-04 by MasterPlexus