j-kaltes/Juggluco #9: Mirror: allow hostname instead of IP

j42c, 2023-03-09:
To deal with changing IPs and DynDNS, it would be great if hostnames were allowed instead of IP addresses.

j-kaltes, 2023-03-09:
That is now not possible and also not easily changed. You can only use multiple IPs.
Maybe some other solution is possible.
What is exactly your situation?

j42c, 2023-03-09:
I'm running juggluco-server behind a DSL access to collect the libre3 data. There I have no static public IPv4 address, it changes from time to time. If hostnames can't be used, I think the next best solution would be to use a cloud vm with a static ip for this job or just to forward the connection from there to my DSL access. Or do you have another idea?

pachi81, 2023-03-09:
What about using a VPN connection, so you can use your local IP from everywhere.

j-kaltes, 2023-03-09:
Why not add the IP every time it changes (a few seconds work). It is said that the propagation time of DNS changes can take up to 72 hours (https://ns1.com/resources/dns-propagation), although in my experience it is nearly immediate.

j42c, 2023-03-11:
Thanks for the suggestions!

closed 2023-03-11 by j42c