j-kaltes/Juggluco #10: Scan error 19 on lineageos 20

thanasistrisp, 2023-03-15:
At custom rom  (lineage) and nfc working (e.g. with google pay) when I try to scan libre 1 sensor I get error 19.

j-kaltes, 2023-03-17:
ScanError 19 is caused by an exception. Could you, before you scan, make the following command from you computer?

`adb shell logcat --pid `adb shell pidof tk.glucodata` > logcat.txt`

When everything goes well, the exception will be logged in logcat.txt.

thanasistrisp, 2023-03-17:
> ScanError 19 is caused by an exception. Could you, before you scan, make the following command from you computer?
> 
> `adb shell logcat --pid $(adb shell pidof tk.glucodata) > logcat.txt`
> 
> When everything goes well, the exception will be logged in logcat.txt.

Here is the logcat:
[logcat.txt](https://github.com/j-kaltes/Juggluco/files/11000406/logcat.txt)

j-kaltes, 2023-03-17:
What version of Juggluco do you use?

thanasistrisp, 2023-03-17:
> What version of Juggluco do you use?

4.10.4 in android 13 (lineageos 20)

thanasistrisp, 2023-03-17:
> What happens when you use this one?
> 
> https://drive.google.com/file/d/1dmvlpEskw0CI5vdkN40PVg7mTBJuymZD/view?usp=share_link

Data processing error... try again
[logcat.txt](https://github.com/j-kaltes/Juggluco/files/11001878/logcat.txt)

j-kaltes, 2023-03-17:
I leave it so as it is. Previously it did work with Freestyle Libre 1 sensors. I am currently using a Freestyle Libre 3 sensor. Am not going to change anything and risk making it unusable with Freestyle Libre 2 sensors.

thanasistrisp, 2023-03-17:
> I leave it so as it is. Previously it did work with Freestyle Libre 1 sensors. I am currently using a Freestyle Libre 3 sensor. Am not going to change anything and risk making it unusable with Freestyle Libre 2 sensors.

which app version worked for version 1? Also in readme you say it intends to work for v1...

j-kaltes, 2023-03-17:
The last Freestyle Libre 1 sensors I used was in April 2021. But I heard of others using Juggluco with Freestyle Libre 1 sensors and I haven't changed this part of Juggluco for a long time.
The oldest I can find is 1.3.5 and that is from November 2021 https://appgallery.huawei.com/app/C104993801
I wouldn't try such an old version.

thanasistrisp, 2023-03-17:
> The last Freestyle Libre 1 sensors I used was in April 2021. But I heard of others using Juggluco with Freestyle Libre 1 sensors and I haven't changed this part of Juggluco for a long time. The [...]
so libre1 sensor is end of support? If yes, update readme at home page

closed 2023-05-21 by thanasistrisp