j-kaltes/Juggluco #13: Scan Error using FreeStyle Libre 3

VanessaMGoncalves, 2023-03-20:
I have been using the App and more than once I had problems related to the sensor scan getting a "Scan Error" type error. After trying several times, in most cases I manage to re-establish the connection, but twice I have lost the connection to the App completely and get a message like "Replace sensor?". Why does this happen? Any feedback?

I have been using the following phone: Redmi Note 10 Pro.

j-kaltes, 2023-03-20:
Replace error is what it means, the sensor doesn't work anymore. When you scan with Abbott's Libre 3 app you receive exactly the same error message.

VanessaMGoncalves, 2023-03-20:
But this issue happened more than once, which makes me ask if could be due to the Juggluco App itself.
Sometimes it did not mentioned "replace sensor" but I was not able to reconnect the sensor to the App.

VanessaMGoncalves, 2023-03-20:
Yes! I had it two times already! I have been using some sensors (I tried around 10). 
I also had the Abbott's Libre 3 App installed in all the sensors I tried since this is actually the first step before starting Juggluco App. 

Sometime what I did to try to re-connect the sensor was to go again to the Abbott's Libre 3 App, scan again and after Force the stop of App again. 

Could this be the reason for failing before ending the 14-days of the sensor?

j-kaltes, 2023-03-20:
A "_Replace sensor_" error has nothing to do with running Abbott's Libre 3 app in parallel with Juggluco. It means that somehow the sensor is not working properly, independent of its connection with an app. If you use also the Libre 3 app, you can see that it also generates a Replace sensor error, thus that way you can see that it is not something specific to Juggluco.

But when the Abbott's Libre 3 app and Juggluco are trying to contact with the same sensor at the same time, both will fail. You need to force stop the app you are not using and see to it that it isn't restarted automatically. Also, to get a connection back, you always have to scan the sensor again. So when the sensor was first connected with Juggluco and thereafter you scan the sensor with Abbott's Libre 3 app, you have to scan the sensor again with Juggluco if you want to receive glucose values again with Juggluco (and you have to force stop Abbott's Libre 3 app).

Venipa, 2024-03-30:
Uhm it might be related or not but im also getting a "Scan Error" for my libre3, im unable to scan it with the official app due to the Samsung Galaxy S24 chipset (exynos)
So i've tried it with this app but it shows me a "Scan error" and that i should try it again.
Im unable to use the libre3 sensor as of now, im currently overthinking to switch to another device (pixel or snapdragon chipset device)

j-kaltes, 2024-03-31:
Except when you have a new kind of Libre 3 sensor I have never heard of, you should always  be able to get something different from a **Scan** Error. Did you try again, possibly at a different place on the phone? Maybe you have to turn NFC off and on or reboot the phone, when NFC isn't working properly.

Venipa, 2024-03-31:
> Except when you have a new kind of Libre 3 sensor I have never heard of, you should always  be able to get something different from a **Scan** Error. Did you try again, possibly at a different place [...]

[image]

😐 Still an issue for me 
I've checked online if the serial id is valid and it is

Venipa, 2024-04-02:
> Except when you have a new kind of Libre 3 sensor I have never heard of, you should always be able to get something different from a **Scan** Error. Did you try again, possibly at a different place [...]

i tried scanning an old libre 3 sensor, same error, it seems like its a problem with exynos chipset (maybe different nfc scanner?)