j-kaltes/Juggluco #16: Connection lost

pinggi, 2023-03-31:
After using Juggluco for about 2 days, I moved out of the phone range and noticed that after an hour. 
Juggluco lost connection (wrote No connection, status=133)
I scanned sensor and got this error:

    Data processing Error
    Try again
                      OK

I tried that several times. No change. Then I found out I can disable NFC and enable it again, scan the sensor and don't get the data processing error. Instead a choice to select Juggluco or LibreLink. I chose Juggluco and at the same state. No connection.
I waited if 

I started FreeStyle Libre 2 sensor by first scan with LibreLink app. Then I cose the app.
After an hour I did 2 scans using Juggluco. This started the bluetooth connection after some time.

After using Juggluco for about 2 days I got out of range of my phone and noticed late when I got back to my phone the connection was no longer established.

Juggluco lost connection (wrote No connection, status=133)
I scanned the sensor and got this error:

       Data processing error
       try it again
                         OK

I tried it several times and got the same error. Then I found out I can disable NFC and re-enable it, scan the sensor and it doesn't show the data processing error. Instead I get the option to choose Juggluco or LibreLink. 
I chose Juggluco and I'm still in the same state. No connection.
I was waiting to see if Juggluco would join after some time. But after maybe 20 minutes? the alarm started ringing that Juggluco had no data.

*** No new value since 21:31 ***
                                             CANCEL

Still no connection and the same Data processing error.

I disconnected the sensor and now I don't see its id in the sensor menu.

I had LibreLink installed and not running. LibreLink app had 'Force close' in App menu disabled which means the app wasn't running.

I tried turning off NFC, turning it back on, and then selecting LibreLink. But LibreLink gave an error that the sensor was being used by another application, so I closed LibreLink again.

It looks like the sensor is connected to Juggluco but the sensor turned off bluetooth and Juggluco can't connect to it via bluetooth.

What can I do now? How to re-establish the bluetooth connection with the sensor?

pinggi, 2023-04-01:
If I force close Juggluco in Android App menu, scan the senzor, choose Juggluco as NFC receiver, the app shows "To connect with sensor" notification, but nothing happens then.
Juggluco Sensor screen doesn't show the sensor address, the screen is empty, only "Use Bluetooth" is on.
I have "Sensor via bluetooth" and "Scanning starts app" options on in the Settings menu screen.
```
adb shell logcat --pid 21343
```
Juggluco does't log BT connection attempts, so it probably means it doesn't even try to connect.
Before when I had the sensor address in the Sensor menu in Juggluco I saw these kind of logs
```
3-31 22:45:12.356 14441 14549 W bt_stack: [WARNING:bta_gattc_act.cc(355)] bta_gattc_open_fail: Cannot establish Connection. conn_id=000000. Return GATT_ERROR(133)
03-31 22:45:12.364 14441 14471 D BtGatt.GattService: onConnected() - clientIf=9, connId=0, status=133, address=AE:F
03-31 22:45:12.366 29967 12898 D BluetoothGatt: onClientConnectionState() - status=133 clientIf=9 device=xx:xx:xx:xx:xx:xx
03-31 22:45:12.368 14441 14549 W bt_btm  : btm_ble_disable_resolving_list() rl_state = 0x13, rl_mask = 0x1, to_resume = 1
03-31 22:45:12.368 14441 14549 W bt_btm  : btm_ble_enable_resolving_list() rl_state = 0x12, rl_mask = 0x1
03-31 22:45:12.373 29967 12898 D BluetoothGatt: close()
```

pinggi, 2023-04-01:
If I don't "Force close" Juggluco and try scan the sensor I get

```
   Data processing error
   try it again
                     OK
```

If I "Force close" it before scanning, it will beep, vibrate and show
```
Complete action using
Juggluco   LibreLink
```
I choose "Juggluco", the phone vibrates and shows the notification "To connect with sensor".
Then nothing happens.
I am stuck in this loop.

j-kaltes, 2023-04-01:
Juggluco only works after you have effectively scanned the sensor with Juggluco. For Freestyle Libre 2 this means after you have received a glucose value. So the problem is that you can't scan.
You say you used Juggluco4.12.1_min_android_8.apk, but I have never uploaded a file with that name. Did you get that file from https://www.juggluco.nl/Juggluco/download.html?
From what country do the Freestyle Libre 2 sensors come?
What phone are you using?

pinggi, 2023-04-01:
I downloaded it from http://jkaltes.byethost16.com/Juggluco/download.html
The name was originally Juggluco4.12.1.apk. I added _min_android_8 to know it needs at least Android 8.
Juggluco4.12.2.apk version is there now.
I run it on Samsung S9 Android 10 and the sensor from UK.

> For Freestyle Libre 2 this means after you scanned, you have received a glucose value. 
Yes, it at least beeps and runs the app, but doesn't update the last measurements and doesn't show the sensor address or id on the Sensor screen.
Is it common for the sensor to lock up somehow when it loses connection? I went to the next room and forgot to take my phone with me. And this happened.

j-kaltes, 2023-04-01:
As I said if you can's scan via NFC and no glucose value received via NFC is shown, nothing will work. This has nothing to do with a connection error.

pinggi, 2023-04-01:
Ok and is there a chance to make it working somehow?

I had the sensor for the first time and it has gone bad after 2 days.
I'm wondering if the Juggluco app could have caused this somehow. If it is safe to use this app again next time.
If, for example, the application accidentally wrote something wrong into the sensor, or, forgot to write, or it cannot reset the sensor from the bad state...

Is there a way to see the low level communication, protocol, data and read what's wrong there? I mean from NFC or by physical connection?

I read about xDrip+ and it needs OOP2 for decoding data from the sensor, but Juggluco doesn't need that. Instead it contains native .so libs, but I didn't find any doc that would describe how it works.

j-kaltes, 2023-04-01:
Sensors don't get damaged by scanning. The native libs doesn't contact the sensor. NFC interface is in java.

pinggi, 2023-04-17:
The application sends a custom command to the sensor to enable bluetooth as I understand it. And that's at least one change the app makes.

I found this
https://raw.githubusercontent.com/cryptax/talks/master/BlackAlps-2019/glucose-blackalps2019.pdf
and implemented NFC reader to read memory of sensor to somehow get to know why it failed. I am able to read the sensor id and block data, but it seems they don't make sense. I think it's encrypted and they need to be decrypted first - this is what the native libs do right?

I tried to send "raw read" command as it is described in the pdf. However the sensor returns 0101 = Command not supported.
It seems the pdf describes first version of the Libre sensor and the second version was updated with the security fixes.

Is the decryption process described somewhere or do you understand it? Since I guess you have added the disassembled stuff from the original app to Juggluco I guess...perhaps you could help me what was already found and what is possible to do with the Libre 2 sensor...

j-kaltes, 2023-04-17:
Enabling streaming makes that Juggluco receives glucose values by bluetooth from the sensor. 
This does not influence scanning via NFC and does not create sensor errors.

stephend61, 2023-06-16:
Something I have noticed is if I lose connection I just turn bluetooth off and on again, it continues without issue.
I have also noticed that when the app is open on my phone I am far more likely to get connection errors, since I put the G-watch app on my watch and monitor the levels there I have had very few connection issues if any, they only happen if I am out of range of my phone or I open the app.
I have the samsung zfold3 running android 12 and the samsung watch 3.
I have been very happy with the app so far baring the connection issues. it's been great not having to scan the sensor frequently.  Thanks.

j-kaltes, 2023-06-16:
Do you have Freestyle Libre 2 or 3 sensors and from what country?
Strange that you have these connection errors. My phone has exactly the same chipset, Qualcomm SM8350 Snapdragon 888 5G (5 nm), also with Android 12 and connection errors are not an issue.
More connection errors when Juggluco is in the foreground is very unlikely (You would expect the opposite when battery optimizations are enabled or background activity is not enabled.) I doubt whether you observation reflect a real regularity, or maybe you put Juggluco in the foreground when you have connection errors.
Normally you don't need to turn Bluetooth off and on for Juggluco to reconnect, but it can take a few minutes before a connection is re-established.

stephend61, 2023-06-16:
Hi I have the Libre 2 and from Australia, maybe it's just having the phone on, it disconnects alot if I have the phone on with the app showing with the phone sitting on my desk, so its always in range, but under the same conditions but with the phone off, it seems much more reliable and since I have been using the watch app to monitor it, it doesn't disconnect despite looking at my phone for other reasons.
also I have the latest version of the app 4.19. Moving to the newer versionhasn't made a noticable difference.

j-kaltes, 2023-06-16:
Australian sensors must be started with Abbott's Librelink app or reader. It may still be running in the background and interfering with Juggluco's connection to the sensor.

stephend61, 2023-06-16:
I did force stop the Abbott's app after starting the sensor, I just checked it is still stopped.
all the alarms in the Abbotts app are also turned off.

zecamara, 2023-06-29:
i found juggluco on day 12 of a libre3 use. great app, added xdrip, garmin edge datafield, data logged during bike rides. fantastic.
now that sensor expired, i hsd a libre2 (u.s version) and installed, initialized with stock abbott libre2 app. all good, but no connection to jugglucco. tried terminate the old libre3, but nfc scan in juggluco didnt workn error 18.
tried the patched libre2, which seemed to still read the sensor, but coukdnt quite set the 7777 8888 ports etc succesfully. back and forth pstched and original libre2, now nfc scsns in oroginal libre2 give "sensor ended start a new sensor to check glucose".
is it toast?

before i kill a brand new u.s. libre3, is the simple procedure for the libre3 to use stock libre3, init as normal and just use libreview acct to get the data?  some posts about having to have libreview logged in or not when initializing a libre3 confuse me.

thanks for a great app.  libre3 stock app you cant even zoom plots, horrible app.

zecamara, 2023-06-29:
test123

zecamara, 2023-06-29:
> Australian sensors must be started with Abbott's Librelink app or reader. It may still be running in the background and interfering with Juggluco's connection to the sensor.

i found juggluco on day 12 of a libre3 use. great app, added xdrip, garmin edge datafield, data logged during bike rides. fantastic.
now that sensor expired, i hsd a libre2 (u.s version) and installed, initialized with stock abbott libre2 app. all good, but no connection to jugglucco. tried terminate the old libre3, but nfc scan in juggluco didnt workn error 18.
tried the patched libre2, which seemed to still read the sensor, but coukdnt quite set the 7777 8888 ports etc succesfully. back and forth pstched and original libre2, now nfc scsns in oroginal libre2 give "sensor ended start a new sensor to check glucose".
is it toast?

before i kill a brand new u.s. libre3, is the simple procedure for the libre3 to use stock libre3, init as normal and just use libreview acct to get the data?  some posts about having to have libreview logged in or not when initializing a libre3 confuse me.

thanks for a great app.  libre3 stock app you cant even zoom plots, horrible app.