j-kaltes/Juggluco #18: Receiving data in multiple devices

zibra, 2023-04-10:
Is it technically possible with Libre to receive data on multiple devices? The usecase I'm refering to is as follows: small kid is wearing a sensor and one device is not covering whole House, so we constantly need to change phone locaction to keep it in range. It will be great to be able to use 2 or more devices placed in different locations to extend the covered area.

Additional question is it possible to switch sensor between different devices easily (multiple times during the day) with current version of Juggluco? I mean switching to smartwatch/small smartphone (that kid can carry in thw pocket) when going out for example.

j-kaltes, 2023-04-10:
You can't connect two devices via Bluetooth with the same sensor at the same time and there will always be a region of overlap between two devices and causes interference. But, you can switch between devices.
The easiest way is to wear a WearOS watch that is directly connected with the sensor. 
You can switch between tho phone and the connected watch by setting Left menu->Watch->WearOS config->"Direct sensor-watch connection" on or off.
You can also connect two phones via a mirror connection and turn Stream on a second phone and turn it off in the first phone and turn on receive from in the first phone (Left middle menu->Mirror->Add Connection).
https://www.juggluco.nl/Juggluco/mmol.html#mirror

zibra, 2023-04-10:
Thanks Jaap

zibra, 2023-04-13:
@j-kaltes I've started to configure Watch4 with Juggluco but I can't find `Uploader` option on wearOS version. Is it possible to upload readings from wearOS Juggluco over WiFi/LTE directly to Nightscout?

j-kaltes, 2023-04-13:
No, you have to use a phone as an intermediary, or you can send  it directly via mirror connection to a computer running https://www.juggluco.nl/Juggluco/cmdline/index.html, which can function as a Nightscout server for everything except AAPS. A phone or tablet running Juggluco can have the same function.

zibra, 2023-04-13:
Oh, that's sad ;(

j-kaltes, 2023-04-13:
I didn't put Libreview, Nightscout server and uploader in WearOS version because it increases app size, eats battery power and is unnecessary. Very few have a SIM card in the watch and you can also connect, over the internet, to a phone or tablet running Juggluco or a computer running Juggluco server.

zibra, 2023-04-13:
My use case is to give just watch to my 5yo kid so he won't need to carry phone and I will still be able to monitor his BG over the Nightscout.

To setup alternative options you've mentioned I need static, public IP + ports forwarding on the phone or tablet running Juggluco or a computer running Juggluco server, right?

j-kaltes, 2023-04-14:
Setting up https://github.com/nightscout/cgm-remote-monitor is not easy. Except when you already have a Nightscout server, is it easier to keep a Android tablet or phone running Juggluco at home or to put Juggluco server on  Amazon AWS.

zibra, 2023-04-14:
I have nightscout server already

zibra, 2023-04-15:
So I can confirm now that it work :D I was able to set it up. Galaxy Wear 4 connected directly to sensor, clones data (over LTE) to tablet with Juggluco (with port forwarding, by public, static IP) and Uploader on tablet pushes is to Nightscout.

Thanks for your help @j-kaltes. Great work with Juggluco! 🏅 🙏 

I still think direct upload from the watch to Nightscout will be much simpler setup and less error prone. Hope Uploader module will be available in the WearOS app soon.

ghino18, 2024-08-27:
Hi, i have a same problem with an old man that often leave phone at home and goes outdoor with Lost connection. I see on market smartwatch with sim and full android 8.1 ,some of them state to have NFC. Could a juggluco android app installed on One of them and works as on smartphone?

j-kaltes, 2024-08-27:
NFC on WearOS watches can only be used by Google Pay. Sensors need to be scanned by  Juggluco on a phone and the phone sends the data to Juggluco under WearOS. Thereafter the WearOS version can receive glucose values directly from the sensor via Bluetooth. See https://www.juggluco.nl/JugglucoWearOS/install/manual.html

ghino18, 2024-08-27:
> NFC on WearOS watches can only be used by Google Pay. Sensors need to be scanned by Juggluco on a phone and the phone sends the data to Juggluco under WearOS. Thereafter the WearOS version can [...]

I did not talk about wearOS smartwatches but Watches that use full android 8.1

j-kaltes, 2024-08-27:
WearOS is just Android with some functions missing that are useless on a watch and a lot of functions added that are useful for watches. Android Wear was a better name than WearOS. Like nearly all other smartwatches they have monopolized the use of NFC.
When you use a non-WearOS Android watch,  you have to configure the connection between phone and watch yourself, the same way as when you want to use a sensor with a phone without NFC (see https://www.juggluco.nl/Juggluco/mirror/index.html) and every time when you switch the sensor from phone to watch or back, you have to reverse the mirror direction for Stream values and switch “Sensor via Bluetooth” on phone and watch. 
That is not suitable for people who forget their phone.
What Android watches are you talking about? Harmony or ColorOS watches?

ghino18, 2024-08-27:
I talk about china/AliExpress Watch that are sell with android 8.1 with sim and some of them with NFC and working with PlayStore that seems as little smartphone .

j-kaltes, 2024-08-27:
I don't know anything about those watches. Maybe some of them can be used completely independently of a phone; making use of the NFC of the watch. Some of them have an even more restricted NFC than WearOS: 
https://www.aliexpress.com/item/1005007446422609.html : "_The NFC of this watch only supports unlocking the electronic lock, and does not support the payment function."_
https://www.aliexpress.com/item/1005007166280686.html : _"NFC: only Access Control"_

ghino18, 2024-08-27:
Ok, so the NCF seems to be a type that does not activate sensor