j-kaltes/Juggluco #20: WearOS black screen

zibra, 2023-04-23:
WearOS version quite often start with blask screen and nothing more happens. I need to try multiple times with restarting whole watch to make it work. Is there any way to debug whats wrong?

j-kaltes, 2023-04-23:
I also have that watch and I never experienced this problem. Have you running something in the background that is keeping the watch busy?
What do you mean with black screen? Is the whole watch black, the watch face of Juggluco black, or do you come into a black screen when you go the the Juggluco app?

zibra, 2023-05-01:
Thi is how it looks @j-kaltes 
https://youtube.com/shorts/QUgl8DqAdjw?feature=share

j-kaltes, 2023-05-01:
Yes, that is really something strange I have never seen with Watch4.
A logcat after this problem would be helpful:
`adb shell logcat -d >logcat.txt `
To easier see what comes from Juggluco:
`adb shell logcat --pid '$(pidof tk.glucodata)' -d > logcatpid.txt`
More information can be obtained, by installing a logging version of the WearOS version of Juggluco from:
[https://drive.google.com/file/d/1Log4w14a-F9CTqlBEKAvKNXyLkNaQVYJ/view?usp=share_link](https://drive.google.com/file/d/1Log4w14a-F9CTqlBEKAvKNXyLkNaQVYJ/view?usp=share_link)
After running this version of Juggluco, you can get the logfile with:
`adb shell run-as tk.glucodata cat /data/data/tk.glucodata/files/logs/trace.log > trace.log`
After you have reinstalled the non-logging version, you can remove this logfile with:
`adb shell run-as tk.glucodata rm /data/data/tk.glucodata/files/logs/trace.log`
This all after connecting the watch with the computer and pressing:
`adb connect IPOFWATCH`
IPOFWATCH: the IP of your watch, see: https://www.juggluco.nl/JugglucoWearOS/install/adb.html