j-kaltes/Juggluco #21: Connection or notification problems

Chaboum1, 2023-04-26:
Hi, 
I've been using Juggluco 4.14.1 for a few weeks, it was perfect. 
But for 3 days now: I get no notification with the glucose level: i just have a notification saying "to connect with sensor"   ?? When i press it, it opens the app, I can scan with no problem. But this "blank" notification without value is the problem. What to do to solve it.
I have changed nothing in the settings !! It sees my sensor but .... ???
Am with Libre 2 in France.
Thanks

j-kaltes, 2023-04-26:
What does Left menu->Sensor show?

Chaboum1, 2023-04-26:
[image]
I have the usual info of the sensor

Chaboum1, 2023-04-26:
The values are registered in the graph OK, and  the floating glucose  works.  
But the notification at the top just displays a grey round with no value.    That's strange.