j-kaltes/Juggluco #22: missed source layout/notification_action.xml

MasterPlexus, 2023-05-02:
I've got an error message during compiling. Is here maybe a source missed in GitHub?

`Unable to locate resourceFile (/home/dirk/StudioProjects/Juggluco_new/Common/build/intermediates/merged-not-compiled-resources/mobileLibre3/release/layout/notification_action.xml) in source-sets.`

j-kaltes, 2023-05-02:
As you see in the error message, notification_action.xml is not part of the source of Juggluco. If rebuilding doesn't help, put the source in a new directory or remove some files created by Android studio, for example everything in .idea.

MasterPlexus, 2023-05-04:
Correct. I clean build, and it become better. not sure what I did in between. Sorry.

closed 2023-05-04 by MasterPlexus