j-kaltes/Juggluco #23: Sensor Error

pachi81, 2023-05-08:
Hi,

it is more a question, because I have a lot of problems with Sensor Error while my glucose value is rising or falling very quickly.
As these rising and falling values are expected in such situation, I think the Sensor Error security from Abbott is not correct and could bring me in problems, as there is no new value for about half an hour or more. If I do not scan my bloodsugar in this time, I would get real troubles...
So my question is, do Juggluco get the real glucose value for such readings or only the error code?
If there is also the glucose value, can you please show it next to the Sensor Error information? So I can decide, if the value makes sense or not.
Thank you!

Regards, Michael

j-kaltes, 2023-05-08:
With Freestyle Libre 2 sensors part of the interpretation is done in the Android app, but with Freestyle Libre 3 sensor all interpretation is done on the sensor itself. The app only has to decrypt the value. It is not the case that it receives a value and a reliability in indicator and then only shows a value when it is reliable. It doesn't have a value when it is unreliable. 
On the Juggluco WearOS watch face Juggluco only shows a heart rate when it is reliable, so in principle Juggluco could be less stringent with the heart rate, but that is not possible with Freestyle Libre 2 sensors.
For Freestyle Libre 2 sensors the raw history values are saved in data.dat but not used.

pachi81, 2023-05-08:
Hi,
thank you for the response, I have a libre 3 sensor and if I understand you correct, the Sensor Error is set in the sensor and no data is transmitted!? So there is no way to get the value next to the Sensor Error message...

pachi81, 2023-05-24:
Hi,

just another question, can you also send the error codes as broadcast? So I can react on them.
If possible, can you send them as global broadcast (without specifying an package), then every app, which [dynamically register on broadcast](https://developer.android.com/reference/android/content/Context#registerReceiver(android.content.BroadcastReceiver,%20android.content.IntentFilter)), can receive them without changing the manifest file (like Tasker).

Thank you!

Regards, Michael