j-kaltes/Juggluco #28: Getting data in Libre 3 after installing Juggluco

sshivaji, 2023-05-12:
I want to recommend this app to others, however, some people still want output in Abbot's libre 3 app. I am unable to get output on Abbot's apps and Juggluco simultaneously. Is there a way to do this?

Thanks a lot for the juggluco app. I was diabetic and no longer am, thanks in part to this app. The analytics helped a lot in debugging my blood sugar spikes, lowering my average sugar and much more! You deserve a standing ovation.

MrGutis, 2023-05-12:
There's an patched libre 3 app (14 days version) which has embedded Jugglucco version. You can setup it to act as source and then separately Jugglucco app as an receiver. You can search for "Libre 3 mit Patched App" and find the document with instructions how to set it up.  You'll see the readings in both apps

sshivaji, 2023-05-12:
Thanks a lot! I tried the libre 3 patched app, but when clicking on patched libre broadcast, it says no app listening to librelink. Thus, I cannot send back from juggluco to libre.

Maybe the only way is to make patched libre as the source, and to rescan my sensor and Jugglucco as a receiver. I don't want to do this now as Jugglucco is a billion times better. However, a relative wants to use this app, I will recommend this approach. If this works for someone, I can close the issue. Thanks again.

sshivaji, 2023-05-22:
The most reliable solution for tech users is just to uninstall the libre 3 app after you get the libreview.com (or app) credentials. Then we don't have to worry about bluetooth disturbances. For NON-tech users, I don't know what the best solution is as the installation itself is difficult and the provided patched libre 3 does not work on every region. I will open a separate issue for helping non-tech users.

closed 2023-05-22 by sshivaji