j-kaltes/Juggluco #29: Ability to put in notes for a particular time of the day

sshivaji, 2023-05-13:
I see that we have the ability to put in numbers on any point in the graph followed by activities. However, if you see below, we are restricted to numbers and 6 activities. Can we allow free-form text such as "large apple", "cafe latte" or "boarding a flight"? This makes it easier to search for these events in the log and see how your body reacted.

[image]

sshivaji, 2023-05-13:
Looks like you can add labels for large apple or cafe latte, or food labels, the only thing missing might be a custom event. I guess we can always play with labels and get a behavior similar to what we need.

j-kaltes, 2023-05-13:
You can specify up to 40 labels.  Under the label for carbohydrate, you can also specify a meal out of its components:
[image]

sshivaji, 2023-05-13:
Wow, thats good enough! I will add in custom labels like "light exercise", "in flight" etc. How did you change your app language to Portuguese? :)

j-kaltes, 2023-05-13:
Change phone language you can do in Android settings configuração do sistema -> idioma a região.
Labels you can set in Juggluco Left menu->Definições->Etiquetas numéricas.

sshivaji, 2023-05-13:
I appreciate you changing your whole android system language to Portuguese
to explain! I thought there was a way to change language within Juggluco
itself.

I understood the method to add labels. Will try creating and testing labels
for my use cases. Thanks!

sshivaji, 2023-05-22:
The label solution works well, closing issue to clean up the tracker.

closed 2023-05-22 by sshivaji