j-kaltes/Juggluco #31: Can I get rid of the libre3 app after creating a libreview.com account?

sshivaji, 2023-05-17:
The libre3 official app gets in the way of the bluetooth connection with juggluco. Should I install the libre3 app, activate the sensor, then use juggluco and uninstall the libre3 app. After 2 weeks, I can repeat the above process?

Is there a way to directly activate with juggluco. I much prefer the Juggluco analytics etc, but occasionally struggle with the connection issues. I am technical and can debug the issues via adb if it helps improve the project.

I also see that you have recent changes to the main branch, should I use that version?

j-kaltes, 2023-05-18:
You can activate Freestyle Libre 3 sensors with Juggluco, just scan the new sensor (after giving Juggluco a Libreview Account ID.)
You can also force stop the Libre 3 app. If that doesn't work you can disable it with:
`adb shell pm disable-user --user 0 com.freestylelibre3.app.nl`
In stead of nl you should use your language code. You can also find the package name with:
`adb shell pm list package freestylelibre3`
You can enable it again with:
`adb shell pm enable com.freestylelibre3.app.nl`

sshivaji, 2023-05-19:
Wow, this is amazing! I had no idea that we can activate new sensors with the Juggluco app!! I think a simple solution for those who don't want to use adb is just to uninstall the libre 3 app after you get the libreview account ID setup. I will with the next new sensor and report back.