j-kaltes/Juggluco #35: New Libre 3 sensor pairing process

sshivaji, 2023-05-27:
I recently changed a sensor. The process was like this for me:

1. Apply new sensor, scan it with Juggluco. It still takes 60 minutes to activate it. I think I should have activated it BEFORE removing the old sensor to not lose one hour of data.

2. I added my Samsung Galaxy watch 4 to Juggluco. I noticed that it's more stable if you leave it for an indirect connection for around 12 hours. After 12 hours, I switch to a direct connection with the Samsung Galaxy watch 4. The readings are quite stable then. I might get a disconnection every day or two, but only for a minute.

Questions:

1. Battery life on the Samsung Galaxy watch 4 is rather limited. I only want Juggluco to work on the watch. What can I do to optimize battery life.

2. Do you have any comments on the above setup?

j-kaltes, 2023-05-27:
Yes, one can better activate the sensor an hour before the old sensor ends. Juggluco can be used with two sensors in parallel, so a little overlap is no problem.
I have not noticed that after 12 hours the connection between watch and sensor is better. In my experience, one can connect a Freestyle Libre 3 sensor from the beginning directly with the watch.
The best way to preserve the battery is to run as little apps as possible. So uninstall, disable or force stop apps you don't use. Don't light up the light of your watch unnecessary, for this you can turn on _power saving_.

sshivaji, 2023-05-27:
Thanks for the tips! Next time I will attach 2 sensors for that dreaded hour. I did not realize that Juggluco supports 2 sensors at once. 

I had the issue that when I tried a direct watch connection at the beginning, it was losing the connection many times. Maybe I will do some adb debugging on the watch with logs to help?

I will try more things on the watch to optimize battery life based on your suggestions. Brightness and also some other settings seem to drain battery. How much battery life are you getting now on the Samsung Galaxy 4?

j-kaltes, 2023-05-27:
It now says 44% available 10 hours and 14 minutes with Battery saving, so the full battery would last a little more then 23 hours.  Without it is 8 hours 32 minutes left, so a little less than 19 hours for a full battery.

sshivaji, 2023-05-27:
Wow, I have similar numbers. I was hoping it would last 2 days without
charging. I will try more experiments.

The other option is to use a watch that lasts for a few weeks without
charging, but the only way to get the live numbers will be with android
notifications. These notifications always make a buzz on the cheap watch
sadly.

j-kaltes, 2023-05-27:
One can also connect Juggluco to two WearOS watches.

MiBand/Amazfit watches do a week on a battery. Via Watchdrip can Juggluco send watchfaces containing a glucose value to them, but the watch is a sixth of the time unusable because nothing but "Syncing" is placed on the screen.

sshivaji, 2023-05-27:
Wow, I had no idea that we can do Miband and Amazfit watches with watchdrip. This is very useful potentially setting up for my diabetic father. It's a bit annoying/sad that 1/6 of the time, it won't show the value. With the samsung watch, I always get values.

sshivaji, 2023-06-01:
Will close issue, thanks for the feedback!