j-kaltes/Juggluco #36: App crashes when moving time

LuisPuertollano, 2023-05-29:
Hello,

App crashes when trying to move left (into the past) or right (into the future or to see sensor number). See attached video.

https://github.com/j-kaltes/Juggluco/assets/77344100/166de036-a2fd-4c0a-a425-656a84c3d18b

I'm logged into libreview, receiving data locally from the patched libre3app, and data updates regularly without problems in juggluco...just trying to look into the past or future causes the app to crash. 
Despite the crashes, data is forwarded without delay to xdrip and thrn aaps, so I have to guess there is something in the graphics going on.

Help please? 😇

This keeps happening:
In Android 12 and 13
In Juggluco version 4.13.x, 4.17.x and 4.18.0 (latest)

I guess something is off in my phone, but i don't know what exactly.
[image]
[image]
[image]
[image]
[image]

I can open the search by date option, but when i click on a selected date, the app crashes "softly". As said before...data keeps beeing forwarded and loop doesn't open.

Thanks in advance! 
BR

j-kaltes, 2023-05-29:
You get that sort of problems when not all data is available in Juggluco because you reuse an old mirror connection in the patched libre 3 app. 
The libre 3 app thinks that all data before a certain point are available, but by using an old connection for a Juggluco that doesn't have this data this is not true.
This can be solved by pressing resend data for this connection in the patched Libre 3 app.
Probably you are also using on old patch of the libre 3 app.

LuisPuertollano, 2023-05-29:
This means, dhould I delete the Libre 3 app and take over the connection with juggluco, I should be able kill 2 birds with 1 stone?

j-kaltes, 2023-05-29:
I didn't say that. You could do that, but you don't need to.
If you want to keep to your current configuration, you should press "Resend data" in the Libre 3 app.
But you are right, connecting Juggluco directly with the sensor is the best option.

LuisPuertollano, 2023-05-29:
Ok, "Resend Data" solved the problem, thank you! I´m also wanting to have the less apps the best,..for the sake of simplicity...and the last(s) version(s) of juggluco added the possibility to have a 2nd sensor in parallel for the warm-up period...thats just great :-)
Anyway, thank you again!

closed 2023-05-29 by LuisPuertollano