j-kaltes/Juggluco #37: Custom triggers/alerts for fast rising blood sugar beyond 170 or a custom value

sshivaji, 2023-06-01:
Is there a way to program an alert such as your blood sugar is rising fast beyond 170? I am curious if a text notification is possible via an event or what the other options are. I note that there is a feature to send notifications via the Juggluco app. I could potentially use an android app automator and turn this value on with certain rules. However, I wonder what the best way to do this is.

j-kaltes, 2023-06-02:
Juggluco itself doesn't have a fast rising alarm. One could in principle write an android app that listens to a glucose broadcast that generates these kinds of alarms.

sshivaji, 2023-06-02:
Is the best way to listen to the nightscout broadcast? I guess the
alternative is to have an alarm on a value like 160 or something and maybe
leverage the android notification capability that Juggluco has.

j-kaltes, 2023-06-02:
https://www.juggluco.nl/Juggluco/glucosebroadcast.html

sshivaji, 2023-08-16:
Thanks, that will work!

closed 2023-08-16 by sshivaji