j-kaltes/Juggluco #40: Vertical pinch-to-zoom broken - results in "app not responding"

jaimet, 2023-06-08:
Hi.

Horizontal pinch-to-zoom works (time scale) works "as expected" but vertical pinch-to-zoom (glucose scale) behaves strangely, often resulting in either an  "App not responding" or some kind of "app reset". I have tried this with "Settings/Manually scale/Glucose" both enabled and disabled, but neither seem to function correctly.

Is it possible to have "horizontal pinch-to-zoom" and "vertical pinch-to-zoom" working the same way? (It's really confusing otherwise.)

PS Thank you for writing juggluco. It's brilliant! I no longer have to use a FL2 reader! :partying_face: :tada: :pray:

j-kaltes, 2023-06-09:
As you can read in _Left menu->Settings->Help_ under _Graph range: low-high_, by moving up and down in the upper half of the screen the upper limit is moved up and down and by moving in the lower half, the lower limit is moved up and down. This is much easier than pinch-to-zoom.
For this, _Manually scale Glucose_ has to be checked.

jaimet, 2023-06-09:
> This is much easier than pinch-to-zoom.

I understand and appreciate your opinion. My opinion is the opposite.

Would you consider a PR to add "Pinch-to-zoom on the glucose axis" as a toggle/option, or would I just be wasting my time?

j-kaltes, 2023-06-09:
Shut up!  That is totallly nonsense!