j-kaltes/Juggluco #41: scan error on sensor change

Npavkov, 2023-06-09:
I changed to a new libre2 sensor and it scans reliably in libre2 app, but juggluco always gets a scan error (18). What does scan error 18 mean. is there a trick to getting the sensor into juggloco without scanning?

j-kaltes, 2023-06-09:
From what country is the Libre2 sensor?

Npavkov, 2023-06-09:
I believe it is US. Is there a way to tell?

j-kaltes, 2023-06-09:
On the box there is text in English and Spanish and it includes also an alcohol wipe. https://myehcs.com/product/freestyle-libre-2-sensor/
Today yesterday and today,  I received a lot of posts of users who can't scan **new** US Freestyle Libre 2 sensors with Juggluco. Can you use the patched Libre 2 app? https://www.juggluco.nl/USLibre2

j-kaltes, 2023-06-09:
Do you know the date on the last sensor and the last four sensors?
The dates of the sensors of two people with problems are: 2023-12-31 and 2023-11-03

Npavkov, 2023-06-09:
2023-12-31

j-kaltes, 2023-06-09:
Is that the date of the bad sensor? Do you know the dates of the good sensors?

Npavkov, 2023-06-09:
the good sensors are dated 2023-10-31. they also had english and spanish on top the box.

j-kaltes, 2023-06-09:
https://www.totaldiabetessupply.com/products/freestyle-libre-2-sensor-14-day
Here you can read: 
_FreeStyle Libre 2 Sensor Kit Includes:
Sensor
Sensor applicator
(2) Alcohol wipes
Product information card
Reader not included_

Npavkov, 2023-06-09:
the boxes for the new sensors look EXACTLY the same as old sensors.

Npavkov, 2023-06-09:
I have a collection of my old sensors (even a few that have feel off) and they all report data processing error and scan error 18 on the new version of juggluco. I not sure if that is notable or not.

Npavkov, 2023-06-09:
if I use the libre2 patch in this post, will the sensor still send data every minute to juggluco?

Npavkov, 2023-06-09:
I am unable to get the patch in this post to work on my phone, when I try to scan the sensor with it , it just hangs my phone and I have to restart......

Npavkov, 2023-06-09:
ok, tried the new patch again. Let me ask you a question..... what do you call it when the patch app says "apply the sensor", then "scan the sensor"..... and when I scan it vibrates and makes the NFC sound and afterwards the app does not show any acknowledgement of the scan, and the "back button" and I get a message on my phone "libre2 is not responding"?

Npavkov, 2023-06-09:
yes!!

j-kaltes, 2023-06-09:
The app itself hangs. So you only have to kill it. 
I just scanned a European sensor with it and it worked. Sometimes it also doesn't respond at all, but after a few attempts it works.

j-kaltes, 2023-06-09:
> I do notice on the top of the box, it does contain english and spanish on it, but no alcohol wipe in the box?

Also on the box of British sensors is text in English and Spanish: https://www.britishchemist.co.uk/product/freestyle-libre-2-sensor/

Npavkov, 2023-06-09:
ok, I have tried the patch scan>terminate cycle more than 6 times and still nothing. Patch still not responding and no data in juggluco? I am not so sure the sensor I have is European..I think it is US.

j-kaltes, 2023-06-09:
It is strange that it hangs after the scan. Do you have a very slow phone or what? I never have to terminate it after I a scan. Did you install https://apkfab.com/libre-2/test.freestylelibre2.us/apk?h=c23a18f939207ee342d2054c551274f1b91c69a87f9691ac78f94eb49bc4a056
?
Even when I scan an old sensor it works: It gives a "Sensor Ended" message.

Npavkov, 2023-06-09:
I have no issue with the normal libre2 app scanning the sensor-AT ALL!!  My phone is not what I would consider SLOW- I have A LG V30 phone.

Npavkov, 2023-06-09:
any other thoughts or ideas?

j-kaltes, 2023-06-10:
Yes, maybe you could run a logging version of Juggluco and return the log file to me after you have tried to scan a few times.

https://drive.google.com/file/d/1a2p8ZRWyFcLHQHIz_BAEkAJlWKryqYss/view?usp=drive_link
You can get the log file out of your phone with:
`
adb shell run-as tk.glucodata cat /data/data/tk.glucodata/files/logs/trace.log > trace.log`

and then Email trace.log to me.

After you have reinstalled the non-logging version, you can remove this log file with:

`adb shell run-as tk.glucodata rm /data/data/tk.glucodata/files/logs/trace.log`

Npavkov, 2023-06-10:
I am not familiar with this....where do I enter this command?  on my computer or the phone?

Npavkov, 2023-06-10:
Hope this log helps….Let me know what you find…. Thanks for your help.

 

https://drive.google.com/file/d/1a2p8ZRWyFcLHQHIz_BAEkAJlWKryqYss/view?usp=drive_link

j-kaltes, 2023-06-10:
Where is the log? You can email it to jaapkorthalsaltes@gmail.com

Npavkov, 2023-06-10:
sorry sent it to reply of your email..... I just sent it to  the email you referenced....Thanks

Npavkov, 2023-06-12:
Hi, did you receive the log I sent to you? was it helpful?

j-kaltes, 2023-06-12:
Didn't you receive my e-mail in answer on your e-mail with the log file?
I could find where it went wrong, but don't have enough information to do something about it.
You seem to have US Libre 2 sensors. Was the device Aaddress of previous sensors colored red in Left menu->Sensor?

Npavkov, 2023-06-12:
Thanks for responding…..in answer to your question…..NO, there is no sensor listed under LeftMenu>”sensors”…. I was never able to scan it via nfc in Juggluco to get it to startup the sensor. I initially scanned the sensor in libre2 app and it started up fine, I was then attempting to scan in juggluco to start the sensor in juggluco. It fails all scans in juggluco.

 

If I can offer any other help to you to resolve this I am more than willing to try…..

Npavkov, 2023-06-12:
Something that might be of help, is when I scan it in libre2 app, it buzzes normally once then makes a different sound and successfully scans.

j-kaltes, 2023-06-12:
Do you use https://play.google.com/store/apps/details?id=com.freestylelibre2.app.us ?
Then you know that it is a US Libre 2 sensor.
I found somewhere the apks on the internet, but I am uncertain if they are the right ones. Can you give me the APKS from your device?
One can find their names with:
`adb shell pm path com.freestylelibre2.app.us`
Thereafter you can pull the files from your phone one by one with
`adb pull `_FILENAME_
For example when `adb shell pm path com.freestylelibre2.app.us` returns:
```
package:/data/app/~~VPhFIm1p52GtVTRweCTdWQ==/com.freestylelibre2.app.us-IULLEQmXzFzLozvm62O_Sg==/base.apk
package:/data/app/~~VPhFIm1p52GtVTRweCTdWQ==/com.freestylelibre2.app.us-IULLEQmXzFzLozvm62O_Sg==/split_config.arm64_v8a.apk
package:/data/app/~~VPhFIm1p52GtVTRweCTdWQ==/com.freestylelibre2.app.us-IULLEQmXzFzLozvm62O_Sg==/split_config.xxhdpi.apk
```
One pull these three files (output minus package:) from the device, one by one, with:
```
adb pull /data/app/~~VPhFIm1p52GtVTRweCTdWQ==/com.freestylelibre2.app.us-IULLEQmXzFzLozvm62O_Sg==/base.apk
adb pull /data/app/~~VPhFIm1p52GtVTRweCTdWQ==/com.freestylelibre2.app.us-IULLEQmXzFzLozvm62O_Sg==/split_config.arm64_v8a.apk
adb pull /data/app/~~VPhFIm1p52GtVTRweCTdWQ==/com.freestylelibre2.app.us-IULLEQmXzFzLozvm62O_Sg==/split_config.xxhdpi.apk
```

Npavkov, 2023-06-12:
Files sent to your gmail acct.

j-kaltes, 2023-06-12:
I didn't receive you e-mail. Can't you put the apks on google drive and make it public or so and give me the urls?

You can try out a version of Juggluco that just ignores the error. You never know:
Log version: https://drive.google.com/file/d/1xCX8T2KJHyYFVDvuHHbMU8Y7y72iT1wb/view?usp=drive_link
Release version: https://drive.google.com/file/d/1dsVhHftzJDRKFAGkkz6efI1fatkCdCqd/view?usp=drive_link

Npavkov, 2023-06-12:
Still get scan error with ignore release….now I get error 19

Npavkov, 2023-06-12:
check your gmail acct again, I resent them unzipped.

Npavkov, 2023-06-12:
did you get the files that time?

j-kaltes, 2023-06-12:
No, I didn't. I received an e-mail without attachments.

Npavkov, 2023-06-12:
Ok, I never used google drive, how do I do that?

Npavkov, 2023-06-13:
is there a way you could modify Jugglco to allow MANUAL entry of sensor info in place of NFCC scan to start it in Juggluco?

Npavkov, 2023-06-13:
https://drive.google.com/file/d/1b3YqLlc7T82TEgxzShStmBFuSXScKYAx/view?usp=sharing

https://drive.google.com/file/d/17Juod4TbXuHvlCzfxHEqyAeHJvHxgZeA/view?usp=sharing

https://drive.google.com/file/d/1i-UGQR1jtlONhGtJwUGfnVD6kIfku2ag/view?usp=sharing

 

Thanks for your help

j-kaltes, 2023-06-13:
Thanks, 
The files are identical to those I found at apkaio.com, except that you have split_config.xxxhdpi.apk and at apkaio it is xxhdpi, a smaller screen resolution.
I thought that maybe it was the wrong one, because it crashes instead of giving the message "_Please install this app from Google Play._".

Npavkov, 2023-06-13:
can I fix it to work?

Npavkov, 2023-06-13:
I do not use an UNPATCHED version of libre2 APP. It is the stock version. I only use the APP  to startup the sensors in the past. As I said before the sensor is scanning perfectly in the libre2 APP, and started successfully. I still think something has changed however with the sensor. Did the log I sent earlier show a valid scan of the sensor?

Npavkov, 2023-06-13:
why would I use 2.71 when I have been using 2.7.2 all along (many sensors that juggluco was able to scan in the past!)

j-kaltes, 2023-06-13:
Because it is expected not to work.

Npavkov, 2023-06-13:
the apk from the link above:
You can install and run the unpatched US Libre 2 app https://drive.google.com/file/d/1vPS3djE7m3THbhhQfmsARCvgHGBagGxe/view?usp=drive_link

reports the file is a invalid format?

j-kaltes, 2023-06-13:
You should not just install it because of what I said before. But to install it you go to an empty directory and unzip the file:
`unzip "FreeStyle Libre 2 US_v2.7.1_apkpure.com.xapk"`
Then do:
`adb install-multiple -r *.apk`
You will receive the message:
adb: failed to finalize session
Failure [INSTALL_FAILED_VERSION_DOWNGRADE: Package Verification Result]
Because you have still your previous version installed. Don't uninstall it because of the reasons I mentioned previously.

Npavkov, 2023-06-13:
I can not get the adb commands to work
[image]

Npavkov, 2023-06-14:
is there anything you can do to make this work in juggluco?

Npavkov, 2023-06-14:
Just got this this morning from Abbott:

 Dear Valued Customer,

The LibreLinkUp app* will be updated to version 4.7 on June 14, 2023. This means your current version of the app will no longer be supported and will not receive glucose data from your connections starting June 21, 2023. Please download the latest version to continue receiving glucose data from your connections.

j-kaltes, 2023-06-15:
Why not? I also managed to do it with EU Libre 2 sensors, the previous version of US Libre 2 sensors and Libre 3 sensors. Someone in the USA has already sent me an US sensor and when that arrives I can find out what makes them different. For the previous US Libre 2 sensors, one sensor was enough.

Npavkov, 2023-06-24:
any luck in getting the modified libre2 sensors to scan in juggluco?  do you have any kind of ETA?

j-kaltes, 2023-06-25:
As a temporary solution I have made a patch of the US Libre 2.7.2 app that can send glucose values to Juggluco:
https://apkfab.com/libre-2/com.freestylelibre2.app.aq/apk?h=7189cb584d338767d208c333a24f8ff5770301a76c2b4e2dc7e8504f24020905
Making the connection between the patched US Libre 2.7.2 app and Juggluco is done in the same way as with the Patch Libre 3 app, see: https://www.juggluco.nl/Juggluco/libre3/patched.html
You don't need to uninstall the original US Libre 2 app, but it should be forced stopped to prevent it from interfering with the connection. 
It will overwrite the data allready present in Juggluco. To prevent this, you have to first send the data in Juggluco to the patched US Libre 2.7.2 app and before reversing the direction.

Tell me whether is works or not.

Npavkov, 2023-06-26:
Your temp solution worked fine. Thank you for providing this fix and getting the libre2 sensors usable in juggluco again. 
However, I am now having issues getting my Sony sw3 to connect to USb to sideload google play services via ADB, do you have any ideas you can offer me here?

j-kaltes, 2023-06-27:
https://androidaps.readthedocs.io/nl/latest/Usage/SonySW3.html

Maybe this version of Juggluco does work with new US Libre 2 sensors:
https://drive.google.com/file/d/1hrrYsiLnnfLSRwcrMxrTy3QbLmjPolvR/view?usp=drive_link

Npavkov, 2023-06-27:
I was unable to download juggluco5.0.0 version you referenced. It says the file is in your garbage folder and will not let me get to it.

 

https://androidaps.readthedocs.io/nl/latest/Usage/SonySW3.html

https://drive.google.com/file/d/1hrrYsiLnnfLSRwcrMxrTy3QbLmjPolvR/view?usp=drive_link

j-kaltes, 2023-06-27:
I saw that I had to make some more modifications so I deleted it and also the post you was still able to read. I hope nobody has tried it.
I have now uploaded a new one: 
https://drive.google.com/file/d/1UgGuwDXb_OJMRDWMrYQ_m56VaEOY29Ec/view?usp=drive_link

Npavkov, 2023-06-27:
That version still gives me scan error 18…….any ideas?

 

https://drive.google.com/file/d/1UgGuwDXb_OJMRDWMrYQ_m56VaEOY29Ec/view?usp=drive_link

j-kaltes, 2023-07-08:
Juggluco 5.0.0 for arm64 works with new US Libre 2 sensors:
https://www.juggluco.nl/Juggluco/download.html
Does it also work with old US Libre 2 sensors?

seliestel, 2023-07-21:
Hi, I have a very similar problem as the OP, but with a European Libre 2 sensor. The sensor is part of a batch which I had been using without any issues for a couple of months. Yesterday, when my previous sensor expired, I put on this new one. I scanned it twice with the Juggluco phone app as usual, waited for 60 minutes, and it was scanning values for some time. However, the Bluetooth, after the first handshake, was never operational (Last Success value was empty). I tried disconnecting/connecting Bluetooth and phone several times, but could not make it to work. After a few hours, the NFC scanning function also broke down. Now, NFC scans give a "Data processing error" message. I haven't been able to scan values for the last 12 hours. My sensor is dated 30-4-2023. Is there any solution for this? Thank you for your help.

j-kaltes, 2023-07-21:
The last remark has absolutely no  relation with this issue and should be handled in a separate issue.

[11 more replies]

closed 2024-05-03 by j-kaltes