j-kaltes/Juggluco #43: Feature Request: Bluetooth Proxy

jrsmile, 2023-06-13:
Hi, let me first say i really love the app, it makes my life so much easier.
As it is the first app in line on my phone to communicate with my libre3 and can feed libreview for my doc, i would like to request the possibility to inject sensor data from external sources.

I am currently developing a bluetooth proxy based on an esp32 that sniffs the raw bluetooth frames and forwards them via wifi.
if every room in your house has one of them, you could leave your phone on the charger and freely roam around your house without the hastle to remember to take your phone with you if you switch rooms.

if you like the idea i would be delighted to talk more about it.

Best Regards,
JR

j-kaltes, 2023-06-14:
I think you will be the only one making use of this Bluetooth to WIFI gateway. Others will make use a good WearOS watch or a small smartphone.
If you want it, you have to make it yourself. The source of Juggluco is freely available.

jrsmile, 2023-06-14:
i think so too, i might just compile juggluco-server for arm and use rpi2w in every room. thanks anyway.

closed 2023-06-14 by jrsmile