j-kaltes/Juggluco #46: Connection errors 133 since June update on Samsung Galaxy A53

saveheinen, 2023-06-30:
Hi,

first i would like to compliment you with this awesome app that i have been using for a while now. It was always pretty solid/stable until i installed the June update from Samsung on my Galaxy A53.

I am wondering if more users have reported this? 

The stream is very unstable since the june update was installed, when lucky it works for a few minutes but now i am at a point where it only receives readings 1 or 2 times. After that, error 133, see screenshots below.

[image]

[image]

Some facts:

- there was a sensor active when i installed the update, the sensor was still active after the update and could be scanned using NFC without any problems

- i tried disable/enable bluetooth on my phone, disable/enable "use bluetooth" within the app (from the sensor menu and from the settings menu), rebooted my phone several times, forced a stop of the juggluco app
- i have no Freestyle Libre app on my phone
- battery saving is off for the juggluco app
- i replaced my sensor, didn't help (i'm living in NL and receive my sensors from Mediq Direct Diabetes)

I am glad to help with anything.

j-kaltes, 2023-06-30:
I don't know. Is there still an Android 13 problem with your phone? 
Did you try whether setting _Android 13_ makes a difference?
Did you turn on "_Allow background activity_" for Juggluco?
Maybe there is still a battery optimization option to turn off.

saveheinen, 2023-06-30:
Thanks for your quick reply.

Some more info:

- Android 13 checkbox on seems to make the problem worse, so i keep this off
- Wakelock on/off -> no difference
- Turbo on/off -> no difference
- Battery optimization is turned off
- Allow background activity is turned on
- Disabled/enabled "patched libre broadcast" -> no difference

I find it strange that the problems started immediately after the update. I was "hoping" more people are experiencing this.

saveheinen, 2023-06-30:
Seems more people (at least 1) are experiencing issues with bluetooth devices for diabetic usage after installing the latest update -> https://forum.xda-developers.com/t/samsung-galaxy-a53-reverse-a536bxxu6cwe9-update.4600171/#post-88689603

Unfortunately no answers there (yet).

saveheinen, 2023-07-26:
Issue seems to be solved by the July update on my Samsung A53

saveheinen, 2023-08-25:
Solved by Samsung's July update.

closed 2023-08-25 by saveheinen

j-kaltes, 2025-04-13:
When you look at the date of the last post, you see that it was posted at Aug 25, 2023. It is unlikely that you have the same problem.