j-kaltes/Juggluco #50: App crashes during init of Wear OS connection

pachi81, 2023-07-07:
Hi,
the app in version 4.19.0 crashes every time, if I want to remove an existing mirror to wear or if I want to init connection to Wear OS.

I attached to logs:
[log_2023_07_07_14_47_14.log](https://github.com/j-kaltes/Juggluco/files/11982175/log_2023_07_07_14_47_14.log)

Regards, Michael

pachi81, 2023-07-07:
After re-install the app on the phone, it is working again.
But maybe an exception handling is missing in the app...

j-kaltes, 2023-07-07:
This shows where Juggluco crashed:
`2023-07-07 14:42:43.734 13375 11346 tk.glucodata F libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x10 in tid 11346 (WearableListene), pid 13375 (tk.glucodata)`
That is not an exception, but a segmentation violation in libc. It happens in thread WearableListene, which ordinarily is not included in Juggluco.
To find out the reason you should look in the full logcat (not specific for juggluco's pid) under the line  `2023-07-07 14:42:43.734 13375 11346 tk.glucodata F libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x10 in tid 11346 (WearableListene), pid 13375 (tk.glucodata)`  for lines of the following format:
`.... F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***`

pachi81, 2023-07-07:
Ok, unfortunately the logcat history stops later and at the moment I can not reproduce. 
So I have to wait for the next crash...

j-kaltes, 2023-07-07:
Did you add WearableListene to Juggluco?

pachi81, 2023-07-08:
What do you mean? I'm using your app without modifications.

j-kaltes, 2023-07-08:
As I said Juggluco has no thread named "WearableListene". 
If you don't generate to many segmentation violation, you can still get information about the crash by using:
`adb bugreport filename`
This generates a zip archive named filename.zip
In it is contained a directory called FS/data/tombstones
Each file therein contains information about a crash.

pachi81, 2023-07-10:
Yes, the files are still available. I would say, all thombstones from 7.7.2023 are related to this problem.

[bugreport.zip](https://github.com/j-kaltes/Juggluco/files/11999106/bugreport.zip)

j-kaltes, 2023-07-10:
I think that Android generates _WearableListene_ to handle messages. Something in Java_tk_glucodata_Natives_setmynetinfo went wrong. I will look at it later.

pachi81, 2023-07-10:
Yes it was really strange, but after re-installing it, all is working well at the moment.

j-kaltes, 2023-07-10:
I could generate a crash by pressing "Init watch app" and that is solved in Juggluco 5.0.2. I don't know whether you had exactly the same problem.
https://www.juggluco.nl/Juggluco/download.html

pachi81, 2023-07-13:
Ok, it seems to be better.
I close this, because it does not occurs again

closed 2023-07-13 by pachi81