j-kaltes/Juggluco #54: No accurate conversion from mg/dl to mmol/l

pachi81, 2023-07-17:
I have detected, that the conversion from the raw value in mg/dl to mmol/l is not accurate.
For example 145 mg/dl will be shown as 8.1 mmol/l, but it should be 8.0 mmol/l.
It seems you are using 18 as divisor, the correct one is 18.0182 or you can use 0.0555 as factor as decribed [here](https://www.mdapp.co/blood-sugar-conversion-calculator-71/).

Regards, Michael

j-kaltes, 2023-07-17:
Juggluco uses 18 as a conversion factor, just like almost everyone else. Now I see that there really is a glucose value where it makes a difference.

j-kaltes, 2023-07-17:
https://drive.google.com/file/d/1_dC-rGFnbSkc3LTRPZXMprhvs9C6FpUH/view?usp=drive_link
contains a version that uses 18.0182 instead of 18.
When I exported my stream data in mmol/L with that version and compared with the previous version, in 1 of 1481 values it made a difference and it is always such a small difference as shown above. It has the disadvantage that you need to use floating numbers instead of just 18.

pachi81, 2023-07-17:
Is it such a big disadvantage? The result is also a floating number, so it is only the conversion!?
But I don't know your code.
I'm also fine with 18 as it is not a big thing as you mentioned! I only recognized the small difference, while comparing with nightscout, but it never disturb me!
So if you prefer to let it as it is, I'm fine with!

j-kaltes, 2023-07-17:
I thought it was an unnecessary precision. It is easier when steps of 18 are exactly one mmol/L.
180.182, somehow become 180.182007 (I use conversion from mg/L):
#include <stdio.h>
int main() {
	float get=180.182f;
	printf("%f\n",get);
	}
The following is an update, related to this:
https://drive.google.com/file/d/14NOUX0wqXYTnWWoG-331GssJ9cQ_Nd3k/view?usp=drive_link

pachi81, 2023-07-17:
As I wrote, using 18 is also ok. This precision is good enough, at least if it causes more problems.
I thought, it is only a small conversion function.
And I think nobody regocnized the small difference or nobody have a problem with.
So we can close this issue, if you like!

closed 2023-07-17 by pachi81