j-kaltes/Juggluco #57: Problems with bluetooth connection

Geniy83, 2023-07-20:
Hello!
Many (15 people) Belarusian users of Juggluco and owners of Samsung smartphones contacted me. After updating the system, everyone had a Libre 2 Bluetooth streaming failure. And it's impossible to set up a connection, which I just didn't do. Below I give information on the update itself, after which the problems began, and information on the smartphone (one of them).
1. System update (Last from 2023-06-01) - https://doc.samsungmobile.com/SM-A336B/CAU/doc.html
2. Device name - Samsung Galaxy A33 5G (this is one example, all problematic users Samsung)
3. System version - Android 13

lapmat, 2023-07-20:
Hello, I think it's the same issue than #46

Geniy83, 2023-07-20:
Thank you. Yes, this is the same problem (((. The same problem also affected OOP2

aptash, 2023-10-10:
I have the same problem with a new phone "Infinix Zero 20" (XOS launcher 13, Android 13).