j-kaltes/Juggluco #59: Sensor not scanning or connecting with Bluetooth: "Data processing error"

seliestel, 2023-07-21:
Hi, I have a problem with a Libre 2 sensor. The sensor is part of a batch which I had been using without any issues for a couple of months. Yesterday, when my previous sensor expired, I put on this new one. I scanned it twice with the Juggluco phone app as usual, waited for 60 minutes, and it was scanning values for some time. However, the Bluetooth, after the first handshake, was never operational (Last Success value was empty). I tried disconnecting/connecting Bluetooth and phone several times, but could not make it to work. After a few hours, the NFC scanning function also broke down. Now, NFC scans give a "Data processing error" message. I haven't been able to scan values for the last 12 hours. My sensor is EU and dated 30-4-2023. Is there any solution for this? Thank you for your help.

j-kaltes, 2023-07-21:
Can you scan the sensor with Abbotts Libre app?

seliestel, 2023-07-21:
No, I just tried. I cannot.