j-kaltes/Juggluco #60: notification message text color in dark mode

source2k, 2023-07-22:
I got a OnePlus Nord2 phone and activated automatic switching into dark mode after sunset. 
The notification message glucose value is black text on white background at day.
In dark mode, it is black text on black background.
Is it possible to detect the dark mode and change the  notification glucose value text color to white in this case?

j-kaltes, 2023-07-22:
Currently on my phone when I switch between dark- and light mode, the existing notification has the wrong foreground color, but later notifications have a contrasting foreground color. 
So I myself don't have your problems.
Do you keep a non-contrasting foreground color when you restart Juggluco?

source2k, 2023-07-22:
I tested the different situations.
Switching the modes with running app leads to the contrast problem. Waiting for a glucose value change does not help.
After restarting Juggluco the colors are correctly contrasting in dark and light mode.

Hence the easy solution is to disable automatic display mode switching.

I have a similar problem with the always on display (AOD). The status message icon is never updated and shows always the same  glucose value of the time when the AOD activated. Phones of other manufacturers don't seem to have this problem. This is a pity, since I bough the phone with AOD to have the glucose value always displayed.