j-kaltes/Juggluco #61: No voice on list

kowaldemar641, 2023-07-25:
A install juggluco and i have't voice to speak glucose value

j-kaltes, 2023-07-25:
Set Language of Juggluco to English and read Left middle menu->Talk->Help.

kowaldemar641, 2023-07-28:
It's not working. I have not voice to select

j-kaltes, 2023-07-29:
Did you read the following in Left middle menu->Talk->Help?
_On some phones the Android settings for text-to-speech have to be changed for it to work. Sometimes the data of a language has to be installed or the configuration changed. On Samsung phones "Google text-to-speech" has to be set instead of "Samsung text-to-speech engine" in order to choose from different voices. These options can be found by searching for text-to-speech in Android settings or going to Language and input (->Advanced)->Text-to-Speech. Here you can select "Speech Services by Google" and install voice data. In other phones it is under System settings->Accessibility->Vision->Text-to-Speech settings._

kowaldemar641, 2023-07-29:
Yes, i have samsung phone. When i set speech Services by Google juggluko speak polish... THANKS

bobnil, 2024-09-04:
I can confirm if worked when changed the voice engine from Samsung to Google.

1. Go to your phone setting, search for text-to-speech
2. If the preferred engine is Samsung, press the "Preferred engine" header, not the settings symbol.
3. Change to Googles engine
4. Restart your phone or force stop Juggluco for it to recognize the change.

Now the voice list should be populated with available voices.