j-kaltes/Juggluco #63: Show 24 hour Time in Range percentage somewhere prominently

brucestephens, 2023-07-31:
Feature request, really. One nice thing about LibreLink is that on the Home Screen it shows the 24h time in range. That would be nice to add to Juggluco (for example above or below the current reading). I doubt it makes sense to show it in the notification (or on the watch, etc.) because it's not _that_ valuable.

(I mention it largely because it's something LibreLink in the UK has just dropped for some reason leading to lots of complaints because lots of us value it. It's one reason I haven't updated.)

j-kaltes, 2023-07-31:
Time in range percentage for whatever number of days and end time you want, you can view in the statistics screen: Left middle menu->Statistics.
To get statistics of the past 24 hour, make sure that the current time is displayed on the screen and specify in the statistics screen 1 day.

brucestephens, 2023-07-31:
I know the information is available. I'm just suggesting showing it prominently (as LibreLink 2.8.4 and earlier do) is a nice feature. And there's plenty of space above and below the BG value and arrow.

j-kaltes, 2023-07-31:
I don't think it is very useful to display constantly a summary statistic that changes very slowly with time.
You can better concentrate at the current curve and only now and then, when evaluating the overall performance, look at these summary statistics.

brucestephens, 2023-07-31:
Fair enough. I agree it changes slowly, and likely my suggestion of where it could be placed doesn't really fit. LibreLink puts . it on the home page (so ordinarily you see it as you open the app or a few seconds after scanning), so that doesn't really fit with my suggestion. Perhaps there isn't a suitable place where it could sensibly be placed.

While it does change slowly, it's nevertheless a nice quick indication of how things have been going recently. Not really actionable like the BG value and arrow (and graph). But it is (or was, since it's gone in the UK) a figure that's surely trivial to compute and of value (perhaps surprisingly so; I'm not sure I'd have predicted that it would be this useful before having it for a few years).