j-kaltes/Juggluco #67: High memory usage

ah7675, 2023-08-13:
Phone: Pixel 4a (Android 13)
Juggluco: 5.1.5 arm64-v8
Sensor: Libre2 CA

[image]
[image]
[image]

Maximum reached 2.1G if change scale to 24hours.
My phone has 5.9G total memory, 2.1G is about 35%. And sometimes juggluco restarted itself (not sure whether caused by OOM or not).

Is this expected? If not, please guide me how to debug. I have few years experience on embedded Linux but none of Android.

Thank you.

j-kaltes, 2023-08-13:
Your memory usage is absolutely abnormal. When I look at developer options->Memory I see the following:
[image]
[image]

Juggluco also runs easily on old phones and watches with little RAM. Have you done something special that could explain your excessive memory usage?

ah7675, 2023-08-13:
> Your memory usage is absolutely abnormal. When I look at developer options->Memory I see the following: [image] [image]
> 
> Juggluco also runs easily on old phones and watches with little RAM. Have you done something special that could explain your excessive memory usage?

I have no idea. Just observed that my phone is laggy and sometimes juggluco restarted. Therefore look around for root cause. Haven't narrow down which operation(s) cause excessive memory usage.

My daily uses: added new amounts (blood, fast/long insulin), NFC scan upon Bluetooth streaming is disconnected.

Also use along with xDrip+ via "Patched xDrip broadcast". And Libreview is enabled with "send amount".

j-kaltes, 2023-08-13:
If you have experience with embedded Linux, you can maybe look at other ways to view the memory usage, like top or ps or raw /proc/PID.
Did you reboot your phone and/or reinstall Juggluco?
What is the memory usage when you send the data to a mirror device (Left middle menu->Mirror)? You can create one on the same phone with  https://www.juggluco.nl/Juggluco/dub or on your computer with https://www.juggluco.nl/Juggluco/cmdline/index.html .
You can also run a logging version of Juggluco (from https://www.juggluco.nl/Juggluco/download.html) and e-mail the logcat out put to me.

ah7675, 2023-08-13:
The link of 5.0.2-second seems not reachable.
Could you please describe the purpose of testing  mirror function? 

The good news is it can be reproduced. Mem usage is double within 6 hours.

Thanks.

j-kaltes, 2023-08-13:
﻿The purpose of testing the mirror function?
1. You can then reinstall Juggluco without loss of data;
2. You can find out at what place it went wrong. For example when the mirror doesn’t use a lot of memory, something that the original did is the cause. Maybe the contact with the sensor. You can now try this out by connecting the mirror with the sensor. Or maybe sending to Libreview. You can try this out by sending from the mirror to Libreview and not from the original. Etc.

What does the following command return:
`adb shell ps '$(pidof tk.glucodata)' -o cmd,RSS`

ah7675, 2023-08-14:
> What does the following command return?
> adb shell ps '$(pidof tk.glucodata)' -o cmd,RSS
Didn't get the chance to capture before it restart itself :(

Will try next time.

> You can then reinstall Juggluco without loss of data;
> You can find out at what place it went wrong. For example when the mirror doesn’t use a lot of memory, something that the original did is the cause. Maybe the contact with the sensor. You can now try this out by connecting the mirror with the sensor. Or maybe sending to Libreview. You can try this out by sending from the mirror to Libreview and not from the original. Etc.

Make sense, I'll try. Could you please help to build 5.1.5 variant? I tried by my self but didn't get luck.
Thank you.

j-kaltes, 2023-08-14:
﻿The advantage of ps is that you can also use it on devices that don’t have the _memory_ among developer options.
On my WearOS watch, I used the following command:
`adb shell ps --pid ‘$(pidof tk.glucodata)’ -o cmd,ELAPSED,RES,RSS,%MEM`
Giving:

```
CMD                 ELAPSED  RES    RSS  %MEM
tk.glucodata     1-21:23:27  73M  74740   5.2
```

With ps -o hell you can get what the options after -o mean:
ELAPSED Elapsed time since PID start
RSS     Resident Set Size (DRAM pages) 
RES     Short RSS
%MEM    RSS as % of physical memory
Thus “tk.glucodata     1-21:23:27  73M  74740   5.2” means that Juggluco on my watch ran 1 day, 21 hour and 23 minutes, RAM usage was  73 megabyte (RES) or 74740k (RSS) that was 5.2% of total RAM.

ah7675, 2023-08-15:
Here is what I got overnight
```
  PID CMD             STIME     ELAPSED    RSS     VSZ
 7744 tk.glucodata    23:58:32    08:31 220440 14962756
```
```
  PID CMD             STIME     ELAPSED    RSS     VSZ
 7744 tk.glucodata    23:58:32 07:43:04 306936 15978224
```

j-kaltes, 2023-08-15:
On my phone, the same command gives:
```
CMD           STIME        ELAPSED     RSS   VSZ
tk.glucodata  22:08:20     11:18:07   168900 7037224
```

ah7675, 2023-08-16:
[image]

```
$ adb shell ps '$(pidof tk.glucodata)' -o 'PID,cmd,STIME,ELAPSED,RES,VIRT,%MEM'
  PID CMD             STIME     ELAPSED  RES VIRT  %MEM
 4413 tk.glucodata    13:24:08 03:16:49 146M  15G   2.6
```
```
$ adb shell dumpsys meminfo -a --package tk.glucodata
Applications Memory Usage (in Kilobytes):
Uptime: 53591657 Realtime: 67840725

** MEMINFO in pid 4413 [tk.glucodata] **
                   Pss      Pss   Shared  Private   Shared  Private  SwapPss      Rss     Heap     Heap     Heap
                 Total    Clean    Dirty    Dirty    Clean    Clean    Dirty    Total     Size    Alloc     Free
                ------   ------   ------   ------   ------   ------   ------   ------   ------   ------   ------
  Native Heap    58343        0     1328    58332       76        0   487733    59736   558692   547020     7152
  Dalvik Heap     3885        0     1228     3796      376       16     1656     5416    27784     3208    24576
 Dalvik Other      658        0     1332      632       16        0      600     1980                           
        Stack      764        0        8      764        0        0      212      772                           
       Ashmem    17201        0      904    17004       12        0        0    17920                           
      Gfx dev     2840        0        0     2840        0        0        0     2840                           
    Other dev        8        0      296        0        0        8        0      304                           
     .so mmap     3664      144     1224     2972    17140      144      392    21480                           
    .jar mmap      650        0        0        0    24356        0        0    24356                           
    .apk mmap       13        0        0        0      328        0        0      328                           
    .ttf mmap        5        0        0        0       52        0        0       52                           
    .dex mmap      273      264        0        4      424      264        4      692                           
    .oat mmap      287        0        0        0    15124        0        0    15124                          
    .art mmap     1820       20    11620     1516      920       20      432    14076                           
   Other mmap       86        0       20       40      440       28        0      528                           
    GL mtrack      384        0        0      384        0        0        0      384                           
      Unknown     1290        0      324     1284       44        0      341     1652                           
        TOTAL   583541      428    18284    89568    59308      480   491370   167640   586476   550228    31728
```
Although RES show as 146M, but Android developer option display as 7xxM (3h avg).
After drill down by dumpsys, I observe that Native Heap - SwapPss increased from 272235k to 487733k in 40mins.

Full output of dumpsys meminfo:
[Juggluco-meminfo.txt](https://github.com/j-kaltes/Juggluco/files/12357476/Juggluco-meminfo.txt)

j-kaltes, 2023-08-19:
﻿I have checked very new in the native part of Juggluco and I found some missing delete’s. In Juggluco 5.1.6 I have added them. https://drive.google.com/file/d/1wWoKaqwzrX_jB0vIQWag9EPCFmkHiPQ3/view?usp=drive_link
Your  dumpsys meminfo seems to suggest that the Native Heap is the problem, but the memory usage of Juggluco seems to differ wildly from device to device. The memory usage of my c++programming will be the same, so I doubt whether this will solve the problem.
Your dumpsys meminfo differs little from the unproblematic memory usages on my phone:
```
Applications Memory Usage (in Kilobytes):
Uptime: 182021116 Realtime: 405465105

** MEMINFO in pid 23506 [tk.glucodata] **
                   Pss      Pss   Shared  Private   Shared  Private  SwapPss      Rss     Heap     Heap  
   Heap
                 Total    Clean    Dirty    Dirty    Clean    Clean    Dirty    Total     Size    Alloc  
   Free
                ------   ------   ------   ------   ------   ------   ------   ------   ------   ------  
 ------
  Native Heap   189688        0     1108   189676        0        0   142244   190784   398848   375237  
  23610
  Dalvik Heap     4430        0     3208     4352     3716        4       57    11280     7881     3941  
   3940
 Dalvik Other     1401        0      836     1256        4        0       95     2096                    
```
     
Maybe when you run your phone for a longer period some other source of memory usage will turn up.

ah7675, 2023-08-20:
Setup: Both Juggluco are on the same phone, mirrored via 127.0.0.1
```
Libre2 -> Juggluco 5.1.7 -> mirror -> Juggluco 5.1.5 dub (build by my own with LeakCanary) 
                                                                +--> xDrip patched broadcast
                                                                +--> Libreview
```
Both Juggluco show high memory usage.

My observation and theory:
When I just woke up, it seems only increased about 100mb overnight. After that, I have did some log/scans/view as daily uses, then men usage increased about 400-500mb.
I suspect there are some bitmap objects implicitly allocated and cached in native heap. Could be either Android or OpenGL.
But that just my theory, I'm not sure how to prove or diagnose.

[juggluco-mem-5.1.5.txt](https://github.com/j-kaltes/Juggluco/files/12387722/juggluco-mem-5.1.5.txt)
[juggluco-mem-5.1.7.txt](https://github.com/j-kaltes/Juggluco/files/12387723/juggluco-mem-5.1.7.txt)

BTW, how should I pull trace.log from my phone? I don't have root permission.
And the version code and package version are still 410/5.1.5, only version name has changed to 5.1.7.

j-kaltes, 2023-08-20:
From https://www.juggluco.nl/Juggluco/download.html
you can get a version of Juggluco with the debuggable flag set. This makes that you can use run-as, see:
https://github.com/j-kaltes/Juggluco/issues/41#issuecomment-1585617506

Later you have to install again the regular version of Juggluco making use of the -d option of Juggluco, as described at the download page.

I also added some delete's of graphical library functions:
https://drive.google.com/file/d/1yCuq7cEUsAusdiBeEZBJaWaxOl2Vxpbl/view?usp=drive_link

ah7675, 2023-08-22:
Last 24hrs, avg: 89MB, peak: 140MB.
Looks like issue has fixed.
Could you please submit patches to report and provide affected versions?
Thank you very much!

j-kaltes, 2023-08-22:
What do you mean? I put some 5.1.8's at Juggluco's download page.

ah7675, 2023-08-22:
I mean push the source code change to GitHub repo.
Please ignore my request if changes are proprietary.

ah7675, 2023-08-24:
@j-kaltes Would you upload fix to Github? That would be very helpful if I can build from source and test/debug my own version based on 5.1.8.
Thank you.

ah7675, 2023-08-29:
Updated to 5.1.9.
Thank you very much!

closed 2023-08-29 by ah7675