j-kaltes/Juggluco #70: Phone app - Bluetooth setting not saved

MartenRenkema, 2023-08-17:
Phone Samsung Android 13
Juggluco 5.1.5 no3
If we set Sensor via Bluetooth , the tick (vinkje) is not saved and there fore no connection to the sensor. 

Any ideas folks? 

Thanks, Marten

j-kaltes, 2023-08-19:
Do you have a watch or phone connected via a mirror connection? "Sensor via Bluetooth" is automatically turned off when Juggluco receives Stream glucose values from a mirror connection. The solution is the change that. If you have a WearOS watch connected to the phone, you should turn off Left menu->Watch->WearOS config->"Direct sensor-watch connection".

MartenRenkema, 2023-08-19:
Thank you, I will check it with my friend. 
Btw, by a mirror connection you mean something like xDrip/Nightscout? 

Thanks, Marten

j-kaltes, 2023-08-20:
No, left middle menu->Mirror (or _Clone_ in some translations).

MartenRenkema, 2023-08-20:
Loud and clear, thank you.