j-kaltes/Juggluco #73: Application is crashing every night

harentius, 2023-08-21:
Hello
Thank you for your work and such an important application.
Unfortunately, companies don't want to make their solution officially extendable by every one.
I understand a reason of such decision, however it brings a lot of sufferings for enthusiasts and people who “Don'tWait”

I am using Juggluco for receiving data from Libre 3.
I use patched Libre App to send data to Juggluco.
I set up all required permissions so Android doesn’t kill the app. I did it the same way I did for AndroidAPS/xDrip.

It works well, but I encounter crashes under some circumstances. 
It crashes every night from 3 till 5 o’clock.
Also it crashes on sensor change, but unfortunately I didn't realize the exact circumstances yet, I will try to collect logs the next time this happens.
With night crash I collected logs though, by downgrading from version 5.1.5 to 5.1.3 debug and running adb during night.

Do logs tell you something?
Thank you!

```
....
08-21 03:09:18.849  4276  4276 I logno   : I/Maintenance onReceive
08-21 03:09:18.858  4276  9442 I cpp     : last()=20
08-21 03:09:18.858  4276  9442 I cpp     : gethist(0)
08-21 03:09:18.869  4276  9442 I cpp     : nrunits
08-21 03:09:18.869  4276  9442 I cpp     : nrunits 55296
08-21 03:09:18.869  4276  9442 I cpp     : blocksize=4096 take=3 used=0 maxp=4608
08-21 03:09:18.880  4276  9442 I cpp     : lastlifecount=19522 Fri Nov 25 06:44:56 2022
08-21 03:09:18.880  4276  9442 I cpp     : 
08-21 03:09:18.881  4276  9442 I cpp     : getinfo()->lastHistoricLifeCountReceivedPos=12
08-21 03:09:18.881  4276  9442 I cpp     : SensorGlucoseData /data/user/0/tk.glucodata/files/sensors/E07A-XX0D2AHDAA6 sensors/E07A-XX0D2AHDAA6/current.dat scansize=614
08-21 03:09:18.882  4276  9442 I cpp     : removeoldstates /data/user/0/tk.glucodata/files/sensors/E07A-XX0D2AHDAA6
08-21 03:09:18.882  4276  9442 I cpp     : last()=20
08-21 03:09:18.882  4276  9442 I cpp     : gethist(1)
08-21 03:09:18.882  4276  9442 I error   : /data/user/0/tk.glucodata/files/sensors/E07A-XX0D2AHDA0Q/info.dat: No such file or directory
08-21 03:09:18.882  4276  9442 I error   : /data/user/0/tk.glucodata/files/sensors/E07A-XX0D2AHDA0Q/data.dat: No such file or directory
--------- beginning of crash
08-21 03:09:18.883  4276  9442 F libc    : Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x1b in tid 9442 (tk.glucodata), pid 4276 (tk.glucodata)
```

[logcat.txt](https://github.com/j-kaltes/Juggluco/files/12402442/logcat.txt)

j-kaltes, 2023-08-22:
You probably re-installed Juggluco and reused the mirror connection in the patched Libre 3 app, so that old data was not send to Juggluco. You can try pressing resend data in de patched Libre 3 app. All data added after you connected the sensor directly with the sensor will be gone. The other option is to re-install Juggluco.

harentius, 2023-08-22:
Thank you for the answer, will try to update the patch.
Does it also explain night crashes?

P.S. I use patched app in order to also send data in Libre View without interrupting how Libre do this. I've seen that Juggluco can also send data to Libre View, but the data is processed (averaged by 5 points)

j-kaltes, 2023-08-22:
Updating will not solve the problem, pressing 'Resend data' in the patched Libre3 app will.

harentius, 2023-08-27:
I upgraded the Libre Patch, and waited for a few days.
No more night crashes, thank you so much for your help.
I hope, the sensor change will also be smooth now.

closed 2023-08-27 by harentius