j-kaltes/Juggluco #75: Dismissing alarm notification does not stop the alarm

wwestenbrink, 2023-08-27:
Using v5.1.8 on Pixel 4a with libre3 sensor.

Amazing work you've done here, started using Juggluco a few days ago with the goal of getting glucose values on my Garmin Forerunner 255 watch.

Really appreciate the fact that Juggluco alarms are re-triggered after a predefined time interval.

A minor issue, at least for me, is that dismissing the Juggluco alarm notification does not actually stop the Juggluco alarm. 

Note that dismissing the alarm notification from the default libre 3 app also stops the libre alarm. This allows you to stop the libre alarm directly from the watch by dismissing the libre alarm notification, without pulling out the phone.

Please let me know if this is intentional and/or difficult to implement, happy to help out :)

j-kaltes, 2023-08-28:
When an alarm on the phone goes off, touching the notification on the phone stops the alarm. 
This notification can be redirected a watch by the companion app of a watch. 
If the notification is re-directed to Galaxy Watch 4, the alarm stops when you press on “Show on phone”, but not when you press on “Clear”.
When this notification is re-directed to a Garmin watch, it does not stop when “Dismiss” is pressed.
I will look at it whether I can change this.

j-kaltes, 2023-08-28:
Does this version solve the problem:
https://drive.google.com/drive/folders/1qysiLYb2e93WBMjC9MYCqEdW-gGDUb2n

j-kaltes, 2023-08-28:
Does this version solve the problem:
https://drive.google.com/file/d/161Qzz0gwusl-AKSW73TOtwFmHncOoSlJ/view?usp=drive_link

wwestenbrink, 2023-08-28:
Thanks for the quick reply. The new version fixes the issue!

aduernberger, 2024-03-27:
I am using a Samsung S10e with libre3 sensor. My watch is also a Garmin Forerunner 255.  Juggluco version 7.6.5.

I get the alarm notification on my watch but without the Option to dismiss it on the watch.
The only option I get is to block notification for the app.

Unfortunately I can't test the linked version since I get "App not installed as package appears to be invalid" error when I try to install it.

j-kaltes, 2024-03-27:
The modification in the Juggluco link (https://github.com/j-kaltes/Juggluco/issues/75#issuecomment-1695449523) is  put in all following version of Juggluco. You can better install the latest version from: https://www.juggluco.nl/Juggluco/download.html , but you should be able to install it.
Maybe this is relevant:
https://forums.garmin.com/sports-fitness/running-multisport/f/forerunner-255-series/307407/discard-button-for-notifications-missing-after-update-for-firmware-12-23

aduernberger, 2024-03-27:
> https://forums.garmin.com/sports-fitness/running-multisport/f/forerunner-255-series/307407/discard-button-for-notifications-missing-after-update-for-firmware-12-23
That's how I try to dismiss the notification. Using the button on the right normally shows two actions "Dismiss" and "Block App". For the alarm it only shows "Block App".

Yesterday I installed the latest version from https://www.juggluco.nl/Juggluco/download.html

About:
Version Code 457
Version Name 7.6.5
arm64-v8a
Build time: 22:19:56 24-Mar-2024

aduernberger, 2024-03-27:
Works if I enable "Separate".

Preface for the stuff below. You can only interact with the last notification received.

For high/low alarms I see two notifications drop down on the watch. Probably the alarm(that can't be dismissed) and then the "normal" notification (half a second later) that has the dismiss option.

When I enable "Notify" and "Separate"  I once again receive two notifications but the second one only shows the "block app" option.

Haven't seen the "Notify" + "Separate" behavior from the normal alarms.

But it works for my use case with just "Separate".