j-kaltes/Juggluco #76: no watching and restart watching

XuefengYang1128, 2023-08-30:
Version name: 5.1.8
Sensor: Libre 3
Android: 13

Does the app also take breaks? I have noticed that sometimes it stops updating, and there are no new values. After a few moments, it restarts. For instance, it stopped at 17:00 and restarted at 17:13.

Should I be concerned about the duration when no values are being offered? Is it advisable to initiate a sensor by rescan?

[logcat.txt](https://github.com/j-kaltes/Juggluco/files/12477349/logcat.txt)

j-kaltes, 2023-08-30:
Normally Juggluco receives nearly all values when the phone is near the sensor. Rescanning Libre 3 is only needed after using the sensor with another app.
Have you disabled all battery optimizations for Juggluco and given it running background permission?
What phone do you use?
What does Left menu->Sensor show?

j-kaltes, 2023-08-30:
The Honor V10 contains a Kirin 970 chipset.
I have a Huawei Mediapad m5 tablet containing a Kirin 960s chipset. This tablet has a very bad connection with both Libre 2 and Libre 3 sensors. 
This even though I switch to manually battery->App Launch (turning on Auto-Launch, secondary launch and run in background). Under "More battery settings", I also turned on "Stay connected when device sleeps" .

XuefengYang1128, 2023-08-30:
[image]

XuefengYang1128, 2023-08-30:
Thank you so much. Now I see the problem😊

XuefengYang1128, 2023-08-30:
I have one final question. Could you provide any recommendations for a phone that would work better with Juggluco?

j-kaltes, 2023-08-30:
My own phone is already 2.5 years old and has good bluetooth: 
It contains the following chipset:
Chipset Qualcomm SM8350 Snapdragon 888 5G (5 nm)
CPU  Octa-core (1x2.84 GHz Cortex-X1 & 3x2.42 GHz Cortex-A78 & 4x1.80 GHz Cortex-A55
https://www.gsmarena.com/realme_gt_5g-10689.php

The NFC is worse than my previous phone, but that is not so important.

XuefengYang1128, 2023-08-30:
Thank you so much😊