j-kaltes/Juggluco #79: Invalid mirror entry entered

Npavkov, 2023-08-31:
Help....I inadvertently added an invalid mirror entry on phone juggluco when trying o enter mirror entry for watch juggluco. Now when I select "mirror" from menu, the juggluco display goes away (to the background) and NO mirror window comes up. Is there any way to clear the mirrors without reinstalling?

j-kaltes, 2023-08-31:
What did you enter? Can you run a logging version from https://www.juggluco.nl/Juggluco/download.html and send the logfile.

j-kaltes, 2023-08-31:
Did you try  turning off WIFI and mobile data and Bluetooth or re-evoking internet permission?

Npavkov, 2023-08-31:
Tried to enter this BUT entered on phone instead of watch…..

j-kaltes, 2023-08-31:
What exactly did you enter in Mirror? I can't reproduce it?

Npavkov, 2023-08-31:
> Did you try pressing on Left menu->Watch->WearOS config->Defaults?

No I decided to clear data and re-setup settings…. Good now.

Npavkov, 2023-08-31:
Screenshot sent I entered the ip of my phone

Npavkov, 2023-08-31:
How do I set up the watch wearos app to receive values from phone juggluco?

j-kaltes, 2023-08-31:
When I engter the ip of my phone nothig happens.

j-kaltes, 2023-08-31:
https://www.juggluco.nl/JugglucoWearOS/install/mmol/index.html

j-kaltes, 2023-08-31:
> Screenshot sent I entered the ip of my phone
I have not received a screenshot.

Npavkov, 2023-08-31:
I was successfully able to get my galaxy watch 4 to collect the sensor data via Bluetooth, however it is not sending data back to my phone……how do I make that happen?

j-kaltes, 2023-08-31:
If you didn't change anything of the connection between phone and watch by hand (that is only made a direct connecton between phone and watch with Left menu->Watch->WearOS config->"Direct sensor watch connection"), you need to do something like pressing left menu->Mirror->Reinit and/or sync on your watch. Maybe turn WIFI off and on on your phone or maybe press left middle menu->Mirror->Reinit on your phone.
Sometimes there is also something with the Galaxy Wearable app on your phone.
See  https://www.juggluco.nl/JugglucoWearOS/mgdL/index.html

Npavkov, 2023-08-31:
OK, this is interesting…I have been struggling for hours to get the juggluco wear app on my watch and get it to display data from the phone (Mainly because I am unable to use the Samsung wear app on my phone to connect to the watch….I can not signin to my Samsung acct because Samsung wear will not allow me on a non-samsung phone.)  HOWEVER, I was successful at getting the watch paired and connected using the wear OS app. After I stopped and restarted the Bluetooth and the wifi connections on the watch, the watch just started collecting the data from the US sensor. (it has been collecting for over an hour now.)  Ike I said though I am unable to see any data reading in my juggluco app on my non-galaxy phone. I tried restarting connections on the phone and watch again, however that does not help to get the watch to send the collected data to my phone juggluco……so this is where I am at now….stuck with watch collecting data from sensor, but not sending data to the phone juggluco…..any help would be appreciated…….

I would like to see the watch send data to phone via wfi or lte internet and have phone juggluco process the data received….

j-kaltes, 2023-08-31:
> I can not signin to my Samsung acct because Samsung wear will not allow me on a non-samsung phone.) HOWEVER, I was successful at getting the watch paired and connected using the wear OS app. 

That is not true, you can use Galaxy Wearable with a non-Samsung phone. I use it with a non-Samsung phone. Just install it from Google Play and start it. Can you use a Galaxy Watch without that app?

Npavkov, 2023-08-31:
I did install the Galaxy wear app from playstore, however when I run it  asks me to signin to my Samsung acct and I click ‘continue’ but it just returns to the same signin to Samsung acct screen. I am successfully able to signin to my acct via we browser so I know the acct works……ANYWAY, does juggluco depend on the galaxy wear app to transfer data from the watch?

j-kaltes, 2023-08-31:
You can't use Galaxy Watch 4 without using the Wearable app and Juggluco depends on it for the communication between phone and watch.

Npavkov, 2023-08-31:
Well, I don’t know how I did it but my phone juggluco was successfully sending sensor data to watch, and like I said, I restarted Bluetooth and wifi on watch and suddenly the watch connected to sensor and started collecting data (rather reliably I would say!). I am amazed really that after you stated that the juggluco wear could NOT connect to US sensors, mine seems connected just fine. I was able to use the galaxy watch 4 watch without the Samsung wear app……

Npavkov, 2023-08-31:
Checkout this utube video link….. shows you how to pair and setup/use a galaxy watch4 WITHOUT using a phone at all!!

 

https://www.youtube.com/watch?v=3Xwfc6pUAyo

j-kaltes, 2023-08-31:
> [...] I am amazed really that after you stated that the juggluco wear could NOT connect to US sensors, mine seems connected just fine. 

As you can read at https://www.juggluco.nl/Juggluco/changes.html (referred to by https://www.juggluco.nl/Juggluco/download.html). Beginning with version 5.1.0, can the WearOS version of Juggluco directly connect with  New US Libre 2 sensors. Earlier it was not possible.

Npavkov, 2023-08-31:
OK, well all that really does not matter to me right now. All that I care about is getting juggluco wear to send data via wifi back to my juggluco on my phone. Is there any help you can give me with that?

j-kaltes, 2023-08-31:
WIFI on WearOS watches is automatically turned off after some time. To communicate via Bluetooth instead, the Wearable App is probably needed. Normally the WearOS app can partly do that. But why prefer the WearOS app? For that shit app you need the Google app (uninstall it with adb pm uninstall --user 0 com.google.android.googlequicksearchbox) that is a nuisance. You can also not change the settings of watch faces in the WearOS app for example the color of the Watch face of Juggluco.

Npavkov, 2023-08-31:
I understand. However I purchased the Galaxy watch 4 JUST to collect  data from the sensor so I would not have to carry my phone with me always. as I stated I was unable to get the phone to pair using the Samsung wear app, that is why I tried the WearOS app  instead. The wear OS app did connect the watch via Bluetooth (as audio device) just fine and that is the only thing I plan to use it for. SO far the watch has successfully been collecting data for over 3 hours without disconnecting at all. Is there a way I can get the juggluco wear app to send the data it is/has collecting/collected back to the phone vis wifi connection to juggluco.? Can I setup a mirror on the watch to do this or even setup a web server on juggluco to do it?

Npavkov, 2023-09-01:
What version of Samsung wear do you use with a non-samsung phone that allows you to signin to your account for Samsung?