j-kaltes/Juggluco #81: Notification stuck

Igor-bubble, 2023-09-21:
### Discussed in https://github.com/j-kaltes/Juggluco/discussions/80

<div type='discussions-op-text'>

<sup>Originally posted by **Igor-bubble** September 21, 2023</sup>
S20FE, Android 13, Juggluco ver# 410, 5,1,5 Libre 2 Canada

I am on my 3rd sensor with Juggluco. Until recently Android notification on the lock screen displayed latest Juggluco reading. Since earlier this week (possibly coinciding with Wear OS update to 5.0), Notification reads "To connect with sensor" . Never happened before, and can't be cleared. I tried to scan, turn off and on Bluetooth, restart the phone, turn notification permission off and on. Short of deleting and reinstalling Juggluco I am out of ideas. Did anyone experience this and was able to fix without loosing the history in Juggluco?  
P.S. I do have Nightscout and Librelink records, but would love to preserve records in Juggluco
</div>

j-kaltes, 2023-09-21:
When it is on your phone, it certainly has nothing to do with a WearOS update. Did you perhaps turn off "Glucose in android statusbar" in left menu->Settings?

nvanderperren, 2023-11-03:
I also keep having "To connect with sensor" notification. Notifications are turned on in the settings.