j-kaltes/Juggluco #85: Making volume configurable as ringtone or alarm

inversion, 2023-10-07:
Hi @j-kaltes - I have a use-case for making usage of alarm vs ringtone volume for high/low alarms configurable:
- I want the sound to only come through connected bluetooth headphones, and not the phone speaker.
- This is the behaviour with ringtone volume, but with alarm volume it comes from both.

I am happy to take a stab at adding this as a menu option and threading it through the app. Any thoughts, or good example commits I should follow?

I already looked at https://github.com/j-kaltes/Juggluco/issues/19 and based on https://github.com/j-kaltes/Juggluco/blob/primary/Common/src/main/java/tk/glucodata/settings/Settings.java#L429 through to https://github.com/j-kaltes/Juggluco/blob/primary/Common/src/main/java/tk/glucodata/Notify.java#L140 it looks like the audio attributes are not configurable right now.

Thanks,
Andrew

j-kaltes, 2023-10-07:
What about the "_Alarm is alarm_" option in left menu->Settings->Alarms. When it isn't set the Ringtone volume is used.

inversion, 2023-10-17:
I found that this option didn't work for the use-case I described, I won't have access to the device for a couple of weeks, but will follow up when I can test it again. Thanks for replying quickly

j-kaltes, 2023-10-17:
It certainly works. Did you press "_Save_" after unsetting "_Alarm is alarm_"?

inversion, 2023-11-07:
I believe so, but I'm going to close this for now as I can't confirm (don't have access to the device anymore).

closed 2023-11-07 by inversion