j-kaltes/Juggluco #86: after v5.1.13, fail to build Juggluco

tigershark82, 2023-10-13:
Up until version [6.0.4](https://github.com/j-kaltes/Juggluco/tree/ecfb46b1259bb8d273389a47e149281171f00297), I successfully build Juggluco. However, after [6.0.5](https://github.com/j-kaltes/Juggluco/tree/44167e338de29b5ce835488d134b936e142d4149) version, I cannot build because of following errors(BTW, I use a MacBook M1 Pro[Sonoma]).

FROM 6.0.5
```
ERROR: /Juggluco/Common/src/main/res/drawable/dialogbackground.xml:2: AAPT: error: resource color/dialogbackground (aka tk.glucodata.debug:color/dialogbackground) not found.
```

After change as following, I can build successfully.
```
<solid android:color="@color/dialogbackground" /> =>  <solid android:color="#000000" />
```

FROM 7.0.0, I met the following error.
```
ERROR: /Juggluco/Common/src/mobile/res/xml/glucose_widget_info.xml:14: AAPT: error: resource drawable/widget (aka tk.glucodata.debug:drawable/widget) not found.
```
I deleted following line in glucose_widget_info.xml because of no resource definition `@drawable/widget`
```
android:previewImage="@drawable/widget"
``

but now I meet the following new errors.
```
/Juggluco/Common/src/mobile/java/tk/glucodata/Menus.java:85: error: cannot find symbol
	View view = flater.inflate(R.layout.menus, null, false);
	                                   ^
  symbol:   variable menus
  location: class layout
/Juggluco/Common/src/mobile/java/tk/glucodata/Menus.java:95: error: cannot find symbol
    var menusviewview=view.findViewById(R.id.menusview);menusviewview.setOnClickListener(v ->{}); 
                                            ^
  symbol:   variable menusview
  location: class id
/Juggluco/Common/src/mobile/java/tk/glucodata/Menus.java:96: error: cannot find symbol
        CheckBox systemuiview=view.findViewById(R.id.systemui);
                                                    ^
  symbol:   variable systemui
  location: class id
/Juggluco/Common/src/mobile/java/tk/glucodata/Menus.java:102: error: cannot find symbol
        var menusview=view.findViewById(R.id.menus);menusview.setOnClickListener(v ->{
                                            ^
  symbol:   variable menus
  location: class id
/Juggluco/Common/src/mobile/java/tk/glucodata/Menus.java:108: error: cannot find symbol
        var watchview=view.findViewById(R.id.watch);watchview.setOnClickListener(v ->{
                                            ^
  symbol:   variable watch
  location: class id
/Juggluco/Common/src/mobile/java/tk/glucodata/Menus.java:116: error: cannot find symbol
        var sensorview=view.findViewById(R.id.sensor);sensorview.setOnClickListener(v ->{
                                             ^
  symbol:   variable sensor
  location: class id
/Juggluco/Common/src/mobile/java/tk/glucodata/Menus.java:121: error: cannot find symbol
        var settingsview=view.findViewById(R.id.settings);settingsview.setOnClickListener(v ->{
                                               ^
  symbol:   variable settings
  location: class id
/Juggluco/Common/src/mobile/java/tk/glucodata/Menus.java:125: error: cannot find symbol
        var aboutview=view.findViewById(R.id.about);aboutview.setOnClickListener(v ->{
                                            ^
  symbol:   variable about
  location: class id
/Juggluco/Common/src/mobile/java/tk/glucodata/Menus.java:137: error: cannot find symbol
        var exportview=view.findViewById(R.id.export);exportview.setOnClickListener(v ->{
                                             ^
  symbol:   variable export
  location: class id
/Juggluco/Common/src/mobile/java/tk/glucodata/Menus.java:146: error: cannot find symbol
        var mirrorview=view.findViewById(R.id.mirror);mirrorview.setOnClickListener(v ->{
[... 76 lines cut ...]
  symbol:   variable date
  location: class id
/Juggluco/Common/src/mobile/java/tk/glucodata/Menus.java:261: error: cannot find symbol
        var daybackview=view.findViewById(R.id.dayback);daybackview.setOnClickListener(v ->{
                                              ^
  symbol:   variable dayback
  location: class id
/Juggluco/Common/src/mobile/java/tk/glucodata/Menus.java:266: error: cannot find symbol
        var daylaterview=view.findViewById(R.id.daylater);daylaterview.setOnClickListener(v ->{
                                               ^
  symbol:   variable daylater
  location: class id
/Juggluco/Common/src/mobile/java/tk/glucodata/Menus.java:271: error: cannot find symbol
        var weekbackview=view.findViewById(R.id.weekback);weekbackview.setOnClickListener(v ->{
                                               ^
  symbol:   variable weekback
  location: class id
/Juggluco/Common/src/mobile/java/tk/glucodata/Menus.java:277: error: cannot find symbol
        var weeklaterview=view.findViewById(R.id.weeklater);weeklaterview.setOnClickListener(v ->{
                                                ^
  symbol:   variable weeklater
  location: class id
```

How can I build version 7.0.0?

j-kaltes, 2023-10-13:
I have now uploaded some more files.

tigershark82, 2023-10-17:
@j-kaltes Thank you heaps.  It works well now~   👍 👍 👍

closed 2023-10-17 by tigershark82