j-kaltes/Juggluco #87: problem running Juccluco 6.04- NO3

Npavkov, 2023-10-19:
For some unknown reason I am unable to get the sensor info to come up on wearos screen (see attached) I can not get the software to automatically have the watch take control of the sensor, OR switch between bluetooth, WIFI, LTE connections automatically. Am I using the wrong version of juggluco to do this?
[image]

j-kaltes, 2023-10-19:
Sensor information is never shown in that screen. When you have a WearOS watch that is connected to the phone, the watch is shown.
[image]

For sensor information, you need to view Left menu->Sensor.

Npavkov, 2023-10-19:
ok, How do I get the watch info to show up then. The watch is paired to the phone?

j-kaltes, 2023-10-19:
What watch do you have? The Juggluco on the phone is the right one. Did you install the WearOS version of Juggluco on your watch? See: https://www.juggluco.nl/Juggluco/download.html
What does the companion app of your watch show?

Npavkov, 2023-10-19:
I am able to manually setup mirror on phone and watch and the watch does receive the values as expected, however that is when I manually set up the mirror.

j-kaltes, 2023-10-19:
Bluetooth and WIFI on both watch and phone should be turned on as described at: https://www.juggluco.nl/JugglucoWearOS/install/mgdL/index.html
What does Galaxy Wearable app show at your phone? Something like:
[image]

You should remove the old mirror connection when you make use of the automatically created connection.

Npavkov, 2023-10-19:
I do not use the galaxy wearable app on my phone because it is not compatible. I setup the watch using a different phone.
BT and wifi are both turned on on the phone and watch. I had this working a few months ago on a DIFFERENT galaxywatch4, but I had to get a new watch because the old one broke.

j-kaltes, 2023-10-19:
It only works on the phone that is connected to the watch making use of the companion app on that phone. 
Out of what do you conclude that your phone is not compatible with Galaxy Wearable?

j-kaltes, 2023-10-19:
What phone do you use? 
Alle the devices I tried it on Galaxy Wearable runs. On a Realme and a Sony phone it also finds the watch. On a Huawei tablet, it runs, but doens't find the Watch. Bluetooth is very bad and Libre 2 and 3 sensors also don't work with that tablet. So the Bluetooth connection with the watch will never work, also not when using the WearOS or Pixel watch app.

Npavkov, 2023-10-19:
I have ALREADY confirmed with Samsung that the LG V30 phone I have IS NOT COMPATIBLE with the wearable app at this time. That is why I needed to use a different phone to initially setup the watch.....

j-kaltes, 2023-10-19:
A WearOS watch without a companion phone has much less functionality.
Juggluco doesn't work well without companion phone:
    • You need to manually configure the connection between phone and watch. 
    • WearOS turns off WIFI after sometime so that Juggluco on the phone doesn't communicate any longer with the watch. 
     
You should have returned the watch or you need to buy another phone.

Npavkov, 2023-10-20:
OK, I was successful in getting the Watch to Take Over direct connection to libre2 sensor by using a standard device pairing application. It seems to work just fine, except I have yet to test if the watch will automatically switch between Bluetooth/WIFI/LTE connections when needed. BUT...........
I ran across a rather intriguing issue in the process:
When I run Juggluco on watch using the Juggluco watchface, I have a pretty steady ALMOST ROCK SOLID connection to sensor. However IF I change to Absolutely ANY OTHER watchface (even a simple clock with no data complications), then Juggluco begins intermittently disconnecting from the sensor, and does not always reconnect. If I switch back to the standard Juggluco watchface, then the connection goes back to ALMOST ROCK SOLID again??
Is this a bug in the Juccluco wear app on the watch?

j-kaltes, 2023-10-21:
If that happens, it is certainly no bug in Juggluco. WearOS kills unused apps to free memory. An app that is in the foreground, for example the watchface, is the last to be killed. 
To make this less likely, you need to go the Samsung Wearable app and in the App info of Juggluco set "_Allow background activity_".

Npavkov, 2023-10-21:
WearOS does not kill the juggluco app, only disconnects bluetooth connection with the sensor.

Npavkov, 2023-10-22:
This issue from what I can see has nothing to do with memory on the watch. **There is > 300 mb available** when the watchfaces are active. This looks like Juggluco just times out or disconnects the bluetooth connection on its own. 
It still remains ODD to me that when the standard Jugglucco watchface is used....There is nearly rock solid bluetooth connection with the sensor. However when ANY OTHER watchface is active, Juggluco intermittently disconnects from the bluetooth connection. Is it possible that Juggluco still tries to update the standard Juggluco watchface even though it is NOT ACTIVE causing some kind of timeout?

closed 2023-11-09 by Npavkov