j-kaltes/Juggluco #92: Broken Link on the web site and export not working.

stephend61, 2023-11-06:
HI, the links below, are broken, is there a new version ? the current one I have which is 4.18 doesn't seem to be working with the latest code, since I updated my sensor.

yet the I am getting readings on my phone Ok, I just can't seems to get the data on my PC anymore, I think it's in the export stage as I am getting files download to the PC in the sensor folder.

This is the command I have been running for months and it's worked fine till now

juggluco4.18.0 -M : mmol/L  -B data_tmp.csv export scans

BOKEN LINKS:
For this you need a little command line program:
https://www.juggluco.nl/Juggluco/cmdline/download/juggluco7.1.5.xz
The source can be found here:
https://www.juggluco.nl/Juggluco/cmdline/download/juggluco-server7.1.5.src.tar.xz

Thanks.

j-kaltes, 2023-11-07:
Yes that link was broken. I have repaired that.
The command you mentioned was never correct and there is also no reason why your current version of the command line juggluco, would start working differently.
To see the command line options, press
```
juggluco -h
```
You will there find:
./juggluco [-d dir] -M : mmol/L
This means that the meaning of -M is mmol/L, not that you should use "-M : mmol/L" as a command line option: To set the unit to mmol/L, you need to use only "-M".

To save the stream value you need to use -B afilename.
In the following I assume that you have renamed the command to juggluco.
For example:
mv juggluco4.18.0 juggluco
or
mv juggluco7.1.5 juggluco
If you want to use the new version.
Thus, a correct command to save the stream values to data_tmp.csv, would have been:
```
juggluco -M  -B data_tmp.csv
```
or simply:
```
juggluco -B data_tmp.csv
```
If you had used -M previously.
If you want also to save scans to afilename, you can do:
```
juggluco -S afilename
```

stephend61, 2023-11-07:
Thanks for the response,  I am  currently having an issue with getting the data down to my pc ( Linux box) it was working till I updated my last sensor so I don't understand what the issue is.

I cleared out all the folders and  recreated the connection but nothing has come down to the pc.
Is there any debug option that I can use to see what is happening on the comms connection?

j-kaltes, 2023-11-07:
What is _the comms connection_?

To see whether you have specified the connection right, look for Juggluco server at:
`juggluco -l`
and in Juggluco app in the mirror menu and see whether the specifications are complementary.
To see at what port Juggluco is listening and what connections there are, you can use
`netstat -np`
To see whether maybe another examplar of Juggluco server is listening at the same port, use
`ps -e|grep juggluco`

stephend61, 2023-11-07:
Hi, I have checked all those and the app is running and I can see that my phone is connecting to the PC but I don't see any real volume of data being transferred, I am using tcpdump where the ip is my phone on my private network.

tcpdump -i eth0 -s0 -nn -vv -l -c 1000  host 192.168.0.120.

When I look at the sensor folder it only has the sensor.dat file in it and none of the other files in the base data folder have changes since setting it up.

I do have the latests version of the juggluco app on my phone and I did update it around the time it stopped working so may be related.

Is there anything that might stop it just downloading all the sensor data again ?
Thanks.

Thanks.

stephend61, 2023-11-07:
another thing to when you run this command it creates to folders

`root@juggluco:/opt/juggluco# ./juggluco -d data -p 8795                <-------------------
Saving in directory dat                        <---------------------------
root@juggluco:/opt/juggluco# ls -la
total 2492
drwxr-xr-x 5 root root    4096 Nov  8 09:44 .
drwxr-xr-x 4 root root    4096 Nov  7 21:38 ..
-rw------- 1 root root       4 Nov  8 09:44 .jugglucorc
drwx------ 5 root root    4096 Nov  8 09:44 dat
lrwxrwxrwx 1 root root      13 Nov  7 21:46 juggluco -> juggluco7.1.5
-rwxr-xr-x 1 root root 2523432 Nov  7 20:05 juggluco7.1.5
drwxrwxrwx 5 root root    4096 Nov  7 21:47 jugglucodat
drwxrwxrwx 5 root root    4096 Nov  7 22:03 jugglucodata
root@juggluco:/opt/juggluco# ./juggluco 
Saving in directory data                        <-------------------
root@juggluco:/opt/juggluco# ls -la
total 2496
drwxr-xr-x 6 root root    4096 Nov  8 09:45 .
drwxr-xr-x 4 root root    4096 Nov  7 21:38 ..
-rw------- 1 root root       4 Nov  8 09:44 .jugglucorc
drwx------ 5 root root    4096 Nov  8 09:44 dat                              <--------------------------
drwx------ 5 root root    4096 Nov  8 09:45 data                            <--------------------------
lrwxrwxrwx 1 root root      13 Nov  7 21:46 juggluco -> juggluco7.1.5
-rwxr-xr-x 1 root root 2523432 Nov  7 20:05 juggluco7.1.5
drwxrwxrwx 5 root root    4096 Nov  7 21:47 jugglucodat
drwxrwxrwx 5 root root    4096 Nov  7 22:03 jugglucodata

`

stephend61, 2023-11-08:
with this command  what do you expect to see ?  should connection show something ?

root@juggluco:/opt/juggluco# ./juggluco -l
Saving in directory data
xdrip/Nightscout webserver turned off

Mirror port 8795
unit: mmol/L
connection:

thanks

stephend61, 2023-11-08:
when using netstat -np  and netstat -ltunp

I so no evidence that juggluco is even listening on any ports. eg.

root@myapps:/opt/juggluco# netstat -ltunp
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 127.0.0.53:53           0.0.0.0:*               LISTEN      94/systemd-resolved 
tcp        0      0 0.0.0.0:8000            0.0.0.0:*               LISTEN      140/python3         
tcp        0      0 127.0.0.1:25            0.0.0.0:*               LISTEN      351/master          
tcp        0      0 192.168.0.15:8080       0.0.0.0:*               LISTEN      142/python3         
tcp6       0      0 :::22                   :::*                    LISTEN      1/init              
tcp6       0      0 ::1:25                  :::*                    LISTEN      351/master          
udp        0      0 127.0.0.53:53           0.0.0.0:*                           94/systemd-resolved 

I can see other apps I have running there, eg the python one.

stephend61, 2023-11-08:
[image]

stephend61, 2023-11-08:
FYI when building with debug on I get this error :

[image]

j-kaltes, 2023-11-08:
Juggluco server runs without difficulty. If you experience problems you use the wrong settings. As I said look in Juggluco at left middle menu->Mirror and in Juggluco server at `juggluco -l` and see that they are complementary. Post a screenshot of the mirror setting and` juggluco -l `and
show the ip of your computer with `ifconfig` or `ip addr show `

Juggluco server compiles without difficulty both on Manjaro Linux 23.0.4 and Ubuntu 20.04.4 LTS running under Windows 10.
If you have difficulty compiling. See that you have the needed libraries and build tools installed. Go to a clean directory. untar the source and make another build directly and in that directory do simply:
```
cmake ../juggluco-server
make juggluco
```
See the README in the source.

stephend61, 2023-11-08:
Hi I finally got it working, I did have it running ubuntu 22.04 in a proxmox lxc and it had been running find for months, then just stopped, so I create a new lxm and updated ubuntu to the latest and start it there and now it's working again.

I am also using your latest build of the liinux app 7.1.5, plus updated my params as you suggested.

I might keep playing around to see if I can work out what happened . I can see when I run nmap from another pc that the 8795 port is closed , so it's not listening for some reason or something is blocking it.

If I work it out I will let you know, it may help someone else in the future.

thanks for your help.

stephend61, 2023-11-08:
Re the cmake I did get it to compile and run locally , I was just pointing out there was some issue if you turn on debug.
Thanks.

j-kaltes, 2023-11-08:
-DDEBUG=ON now also gives this error message.
The previous times I didn't have this problem, maybe I had it turned off all the time.