j-kaltes/Juggluco #104: Suggestion about library

niedbalski, 2023-12-28:
Hi @j-kaltes great work on this tool.

Question, Is there a chance you can export or guide me to extract the bluetooth/nfc communication library in its own so that we can use it independently?

j-kaltes, 2023-12-29:
If you want to write an app for Freestyle Libre sensors yourself, you do you need Juggluco?
If you like to receive glucose values or entered numbers out of Juggluco, you can make use of a glucose broadcast (https://www.juggluco.nl/Juggluco/glucosebroadcast.html) or the webserver (https://www.juggluco.nl/Juggluco/webserver.html) in Juggluco.

niedbalski, 2023-12-29:
> If you want to write an app for Freestyle Libre sensors yourself, you do you need Juggluco? If you like to receive glucose values or entered numbers out of Juggluco, you can make use of a glucose [...]

Yes, indeed I could use both. But given you already sorted the Bluetooth and nfc layers communication I was wondering if I could take that work to make a library that can be used in other use cases as well.