j-kaltes/Juggluco #106: Send amounts with the nightscout uploader

ACWalker, 2024-01-04:
I was wondering whether it is currently possible, or could be made possible to send treatment amounts via the nightscout uploader. 

For example, I would like to add my fast acting insulin treatment scanned from my EchoPen on juggluco, and then be able to send that to my nightscout server via the uploader in Juggluco. As raised in https://github.com/j-kaltes/Juggluco/issues/95 it seems like it is possible to do this with the inbuild nightscout server to be sent to xDrip, could we have a similar "give amounts" option in the uploader?

j-kaltes, 2024-01-04:
No, it isn’t implemented in the uploader to Nightscout. As you said already, you can use the build in [Nightscout web server](https://www.juggluco.nl/Juggluco/webserver.html) for that. You can also run that web server on a computer https://www.juggluco.nl/Juggluco/cmdline/index.html
The amounts are also send to Libreview.

ACWalker, 2024-01-05:
Thanks for your response. I would use the built in web server, but I like to use the actual UI of nightscout as well, so unfortunately would have to build something on top of the APIs exposed.

I currently use my Nightscout server in the cloud as an archive and as a means to see my sugars from my PC and share with others. Unfortunately I haven't got access to any pumps yets, so i'm on MDI.  The reason i'd like to sync insulin is to be able to use the nightscout bolus calculator functionality, add my insulin data to my archive and see the insulin on board lines displayed on nightscout to help with correction/low decisions.

I had that all working when I was using xDrip, but the problems with the slow pollng or double smoothing of libre data in xDrip just drove me nuts.

ACWalker, 2024-01-12:
@j-kaltes just tried the new update and it works a treat! Thanks again for all your hard work, it's definitely helped make things easier for me

closed 2024-01-12 by ACWalker