j-kaltes/Juggluco #107: can not connect the libre3

SimonXuku, 2024-01-07:
I have a watch OPPO watch4 pro it's a standalone android 11 but no nfc.
I use mirror to active the libre2 with my phone and send to my watch, It works fine.
When I use the libre3,It doesnt work. failure code =133.[image]

j-kaltes, 2024-01-07:
Status=133 is crossed out, thus it was a previous error and this error is just a momentary connection error that happens tens of times a day when the connection is very well.
You need to show also the rest of the screen to see what is really happening.
Does the Libre 3 sensor work with the phone and does the watch receive glucose values that way? Can you use it that way with Floating Glucose?

SimonXuku, 2024-01-08:
yes work perfect with the phone,they can transfer glucose value.I wait 1hr ,watch can not show the glucose.also status = 133.
I don't know why libre2 works fine,libre3 can't work. any bluetooth connection difference with libre2 and libre3?
[image]

j-kaltes, 2024-01-08:
The Bluetooth chip on the sensor will be different and the communication over Bluetooth is totally different. 
If you install a logging version of Juggluco and e-mail trace.log, I can maybe see something. (See https://www.juggluco.nl/Juggluco/download.html under "Bug report".) Also make a trace.log when running on the phone, so I can see the difference. Log at least half an hour.  You can better turn off Libreview, the web server and uploader to hide that information.
When using Libre 2 sensors, how long does the watch battery last? The battery is a little smaller than that of Samsung Watch 5 pro, but the battery is said to last 5 days, while the Samsung Watch 5 Pro batter is said to last 3.33 days.

SimonXuku, 2024-01-08:
I can use libre 2 with aaps and pod dash over 40 hrs. I have not only tested libre2 sensor yet.
I'm collecting the report now,I will post it later.

SimonXuku, 2024-01-08:
I sent the log file to you.please check it.
thanks.

j-kaltes, 2024-01-08:
Can you also give a trace.log of your phone so I can see what happens there. I know what did happen on another phone at the same point, but maybe your case is different.

Did you turn off Bluetooth or "use Bluetooth" on your phone? The sensor hangs up a few times. Something it does when something happens to the sensor, like scanning or a connecting with another device.

SimonXuku, 2024-01-09:
I take the phone far away from the sensor and turn off the phone's power. The phone can not connect the sensor. I did not turn off Bluetooth or use Bluetooth on my phone.but libre2 connection is ok.there is no other device connect the sensor.I don't know how is the sensor,this situation also happened  to my friend who did the same test as me. 
I'll post my phone's trace later time. thanks

SimonXuku, 2024-01-18:
I resend my log to your email with phone and watch's log.
Please check it.
thanks.