j-kaltes/Juggluco #108: LibreView RU

Geniy83, 2024-01-18:
Hello Jaap Korthals Altes.
Many people from Russia turned to me for help. Medicine began to issue them Libre 2 on the condition of using LibreView. But they use the Russian server https://libreview.ru. They cannot connect to this server in Juggluco, Juggluco uses https://libreview.com. As I see the code, you can add a connection to https://libreview.ru to the Libreview.java class, dividing the sensors by region (01 and 08). Is this possible or am I wrong?

j-kaltes, 2024-01-18:
Yes, Juggluco doesn’t work with Libreview accounts created with libreview.ru. The following version does work with libreview.ru, when the smartphone country settings is “RU”. This will not be the definite version, because sending to libreview will no longer work for Russians who were sensible enough to take an account at https://libreview.com .
I have also see if there are more country with this problem.

https://drive.google.com/file/d/1FhWhihLCpTlUBH_OM11R9OCgy5o1J-PB/view

Geniy83, 2024-01-18:
Excellent thank you! There are a couple more questions:
1. Can I use Libre 3 with this version and send data to Libreview.com?
2. If I activate the Libre 2 (ru) Juggluco sensor and it breaks, will Abbott see that I did not activate the LibreLink 2 (ru)? Or what is the best thing to do, activate LibreLink 2 (ru) and then switch?

j-kaltes, 2024-01-18:
1. There doesn't exist a com.freestylelibre3.app.ru and using Juggluco with Libre 3 sensors will connect to libreview.com instead of libreview.ru. So if someone has the same password and e-mail with libreview.com and libreview.ru and has Russia as it's country setting, the Libre 3 sensor data will be send to libreview.com and the Libre 2 data to libreview.ru .
2. If you activate the sensor with Juggluco, Abbott will not notice this from the sensor. On the contrary, if you active the sensor with Librelink and later scan it will Juggluco, it is possible to notice from the sensor that streaming was activated by an app not immediately after starting the sensor. This points to a different app than Librelink, but it there are also scenario's whereby Librelink enables streaming again. If you want that Abbott's doesn't notice anything from Libreview, it is also better to use it only with Juggluco and not with both apps. But I don't think that Abbott's doesn't replace sensors when they suspect that someone does use a different app. When Abbott's recognized users of Diabox sending to Libreview, they are cut off from Libreview, but they didn't report problems with getting their sensor replaced: https://service.freestylelibre.be/optiext/optiextension.dll?ID=oSToOf5TbefOoKEMtsG6dC2_HXQa%2BwVq9wk3fwz5m0paVZ4_hsvKHGB1cC%2BODMsFBRM0QA7S0llamItHofBuYzp5ln3_H

j-kaltes, 2024-01-18:
Now you can download a version with a **RU** option under left menu->Settings->Libreview that determines that Libre 2 sensor data are send to libreview.ru.
https://www.juggluco.nl/Juggluco/download.html

Geniy83, 2024-01-19:
Thank you for the work done!

closed 2024-02-25 by j-kaltes