j-kaltes/Juggluco #111: documentation request: reverse change log sort, newest version on top

kdupke, 2024-01-20:
Thank you for working on Juggluco!

I do not exactly know how this repository is linked to/with juggluco.nl, so do not hesitate to redirect me with my request.

https://www.juggluco.nl/Juggluco/changes.html has a change log for Juggluco.

I request to change to order of the entries, so the newest version always is on top.

This change will make it much easier to consume, as it does not need that much scrolling. Many user may be on on of the recent versions so it may fit a single screen on e.g. the phone display.

Thank you for considering this change.

j-kaltes, 2024-01-20:
Attach a keyboard to your phone and press end.

kdupke, 2024-01-20:
It looks like you see the benefit. It would mean not to add a keyboard nor to hit a key for most people.

Not sure if you edit this in a manual way or if the release notes are done by a script. Either way it is a a small change that benefits many.

tyrex1975, 2024-04-15:
Yes, I agree with the OP. Attaching and using a keyboard to navigate to the end is not a user-friendly solution.