j-kaltes/Juggluco #112: Discussion: Make Alarms more life fitting

kdupke, 2024-01-20:
[NB: I am a type 2, so some of my thoughts may not fit type 1]

While an alarm is an alarm, I wonder if there is not more in it to make our life easier.

* predictive alarm by trend
Raise an alarm by using the trend to predict an hypo/hyper level might be reached soon.

Today I have some sort of a mental alarm, when I see the stream showing me that very soon most likely I will hit an Hypo/Hyper. 
Approaching an Hypo means I may take a real blood sample or right away eat some chocolate.

Of course my body reacts in different time on insulin or glucose. So a predictive alarm should have a time window setting.
E.g. run an alarm when an Hypo may approach in 20 minutes.

Such an alarm could have a less intense level of warning, like light vibration only or do not break through a do-not-disturb.

Benefit of this is to avoid the stress level of a traditional alarm. Not only for the user but also for a partner.
Even in case of finally hitting an Hypo/Hypo alarm, it is less stress because someone is mentally prepared for it.

* increasing alarm levels
Create multiple triggers, which can be configured with increasing level.

If I run below 80 I try to predict where I am heading by trend and taken glucose/insulin. But I do not want to set the Hypo level to 80 of course. 

Having multiple alarm level would help me to create a less intrusive warning ahead of time.

Multiple alarm levels might be easier to implement than predictive alarm levels.
Predictive alarm levels may dismiss the need for increasing alarm levels.

* Different alarms on phone status
See also #99 by @DMS4444 
Depends alarm on phone status (vibrate, silent, do not disturb, regular)

Depending on the time of day respective my current status I may react to alarms in a different way, and alarms may be differnet intrusive to my surrounding.

In the opera a regular alarm may help me to avoid a Hypo, but may also take me out of the room.
During day time a full alarm isn't need at first, if it happens in a meeting it might be intrusive to others as well.
During night an alarm must wake me up and my partner if I do not react.

Being able to define an alarm level on the phone status, and an delay till full alarm goes of will help me to avoid issues.

At all, if an alarm is not acknowledged a maximum alarm as configured must be used.

* (not) use of AI
Of course, much of what we do in handling diabetes is around pattern recognition. Which is a perfect fit for AI. Alarms could benefit from this very much - because an AI might recognize a trend alarm isn't needed because there already a treatment has happened which will change the trend very soon.

But because of the importance of alarms at all, I recommend to keep absolute levels for alarm still, just in case the AI isn't not smart enough. 

Thank you for reading till this point. Looking forward your comments.
Of course, getting positive feedback is not equal on getting it done, that is clear to me.

j-kaltes, 2024-01-20:
Make an app that receives a glucose broadcast from Juggluco and implement is yourself:
https://www.juggluco.nl/Juggluco/glucosebroadcast.html

kdupke, 2024-01-20:
I might get this wrong, but your recommendation means
a) you are opposed to the ideas (all, some or one)
b) you don't want to spend your time on this
c) you simply want to help getting a working setup
d) a test by saying no to see if someone keeps the idea spinning 

Of course your recommendation might be a work around.

However, it means to turn a well done solution into a more complex setup. Instead of making an working setup more convenient.

If it simply is a matter of resources but you like the idea, let us know. There might be the one or the other capable of helping coding

kdupke, 2024-01-20:
Just in case, if you prefer this to be moved to discussion, I happily will do so.

closed 2024-01-20 by kdupke