j-kaltes/Juggluco #119: FR: add IOB to glucodata broadcast

pachi81, 2024-02-02:
Please also add IOB to glucodata broadcast.

Do you also think about COB support in Juggluco? 

Thank you very much!!! 

Regards, Michael

j-kaltes, 2024-02-02:
Carbohydrate-on-board and exercise-on-board are also very important, but difficult to determine correctly. Carbohydrate from different sources have different blood glucose curves. I could determine the glucose curve from the ingredients of the meals people enter in Juggluco, but I don’t even know whether there are people who enter the components of meals in Juggluco. I also don’t know where to find the glucose curves of individual ingredients. I know of the glycemic index, which is maybe useful for type 2 diabetics who want to eat meals with the lowest glycemic index, but it misses all the details of the glucose curve. The research on which it is based probably contains more information, but I have no idea how to get raw research data.
Concerning adding IOB to the glucodata broadcast, I have the following to say. I am far from enthusiastic about broadcasts that contain masses of extra information, which have little to do with its main function. It becomes slower and uses more battery power. There is the risk that Android brings the phone or watch to sleep before it is finished because it takes too long. So only what is really needed should be added to the broadcast. I have added it to the pebble web command, because others did so, but maybe I should not have done so.
IOB is something you need to consider when also looking at other measurements, like the glucose curve and previous meals. It is not something to view in isolation.
When I add something to a broadcast, it will stay there forever.

I have now added IOB to the broadcast, only because it is turned off when you don’t set IOB in the settings of Juggluco.

https://drive.google.com/file/d/1Yhav7c6dzyk8_oO2-JPKpputYzybG0rj/view

When asking for it, you need to supply a default value:
```kotlin
private const val IOB = "glucodata.Minute.IOB"
val iob = extras.getFloat(IOB,Float.NaN) 
```
I have modified the examples at: https://www.juggluco.nl/Juggluco/glucosebroadcast.html

pachi81, 2024-02-02:
Hi,

thank you very much!!! Than I can add it to GDH, as my next version will support IOB and COB from nightscout.
At the moment I´m using you pebble interface for the IOB value, so I think for me, there is a higher battery usage for me, requesting the IOB from nightscout.

I haven´t used COB and IOB before and also don´t really understand the calculation in nightscout. But on the first view, it is only carbs per hour and then calculate, but I think it isn´t such simple :-)

So thank you very much für you support!!!
I close it as resolved.

pachi81, 2024-02-02:
I tested the new version and received GlucodataBroadcast including IOB. 
It is working! Thank you.

One remark to the IOB display in Juggluco. The unit for IOB is "U". But as you do not show any units, it is not so important. But you have enough space :-D

j-kaltes, 2024-02-02:
The carbohydrate-on-board of Nightscout is very strange; not something I am willing to emulate.
Now it shows: 295.1g and it is 17:28
And this are what I have eaten after 10:30 as known by the Nightscout server (when I use api/v1/treatments with the server, I get exactly the same carbohydrates.)

2024-02-02-10:48:48     10      Dextro  
2024-02-02-10:48:55     12      Carbohydra
2024-02-02-12:43:00     11      Dextro  
2024-02-02-12:52:00     89      Carbohydra
2024-02-02-13:17:36     56      Carbohydra
2024-02-02-15:34:11     12      Carbohydra
2024-02-02-15:55:29     9       Dextro  

It is in total 199 carbohydrates and you can’t expect carbohydrates from 10:48 to be still active at 17:28.

pachi81, 2024-02-22:
Hi,
after installing version 7.4.6 the IOB support is gone!?
Can you please add this to the release of Juggluco?
Thank you!

Regards, Michael