j-kaltes/Juggluco #124: Segmentation Fault - juggluco-server

kfishermail, 2024-02-11:
I downoded the following files:
 juggluco-server7.1.10.src.tar.xz
 juggluco-server7.3.0.src.tar.xz
 juggluco-server7.4.0.src.tar.xz
 juggluco-server7.4.3.src.tar.xz

I executed the following steps for each file:
 tar xf jugg*.xz
 cd j*r
 cmake -DDEBUG=on .
 make juggluco
 ./juggluco

versions 7.4.0 and 7.4.3 both gave "Segmentation fault":

  pi@linripper:~/jtest/juggluco-server$ ./juggluco
  Saving in directory jugglucodata
  1707620343 125502 setfilesdir jugglucodata null convfactor=180.182000
  1707620343 125502 initVersion=0
  Segmentation fault

versions 7.1.10 and 7.3.0 seemed to work OK
These are the only versions I tested

I am running Debian:
pi@linripper:~/jtest/juggluco-server$ uname -a
Linux linripper 6.1.0-11-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.38-4 (2023-08-08) x86_64 GNU/Linux

Any thoughts?  For my usage I will run 7.3.0

j-kaltes, 2024-02-11:
Thanks for the bug report. I think I have corrected it: https://www.juggluco.nl/Juggluco/cmdline/index.html
You probably had this BUG only the first time you ran the program.

kfishermail, 2024-02-11:
7.4.4 works for me

closed 2024-02-25 by j-kaltes