j-kaltes/Juggluco #128: FR: Broadcasting to mqtt server

Stan-Di, 2024-02-22:
Hello Jaap Korthals Altes!

I kindly ask you to add the function of sending glucose data to mqtt server. It's very convenient and doesn't require Nightscout. This would be very convenient for Home Assistant, oled DIYdisplays etc...

Thank you for the wonderful program!

j-kaltes, 2024-02-22:
Yes, there is the need to communicate between Android clients without the need to setup a server. But mqtt still requires a server that you have to setup. The only advantage above a mirror connection, is that you don’t need to configure something specific for this application, as when putting Juggluco server on a remote computer (see: https://www.juggluco.nl/Juggluco/cmdline).
Because there are probably no mqtt clients, it could be another route for mirror connections. I don’t think that it would be much easier to configure and a lot of work to implement.

Stan-Di, 2024-02-26:
Thank you so much for your answer! I didn't know about this TCP transfer capability. I will sort this out! After quickly reading the document, I realized that I could also run a script for transfer to MQTT on Termux.

Thanks again for your answer! And thanks for the wonderful application with great functionality!

Remendado, 2025-06-27:
mqtt-client should be a great feature to export glucose data to many dashboards like Node Red, Home Assistant, Telegram bots etc. I've been looking for this feature for years.

j-kaltes, 2025-06-27:
I think that when you are not able to use the web server or glucose broadcast of Juggluco to create an MQTT interface, you will also not be able to make use of the MQTT interface when Juggluco would contain it. So what use is it for me to find how MQTT works?
Getting the glucose values out of Juggluco via the web server or glucse broadcast is easy. The only difficulty is the MQTT part and exactly the same knowledge is needed at both sides of the MQTT connection.

NohWayJose, 2025-06-28:
Is there a document that explains how (in enough detail for a non-technical person) to do this with the Web server?

Please don't write MQTT (https://en.wikipedia.org/wiki/MQTT) off as too difficult or inappropriate or not fit-for-purpose. Or anything else, really!. It is designed for IoT, low data rate, asynchronous scenarios (both ends need not be up at the same time), resiliency and ease of use; exposing data flexibly to be used as needed, independent of the server providing the data, so, in-fact ideal for THIS scenario.

If I knew how to code better and could see where in your code the glucose/time/carb and other details are exposed, I'd fork it and import [Paho](https://eclipse.dev/paho/) myself!

Just for reference, I have two scenarios where MQTT would make Juggluco even more useful

1. I want to stream the regular glucose readings to NodeRed (https://nodered.org/) over MQTT (so there's no data loss in the event of patchy comms) and have NodeRed aggregate the data by week, along with my meals, medicine consumption (slow insulin, metformin, etc) and exercise. Once a week NodeRed will e-mail this data (probably as a graph + table) to my diabetic nurse.
2. I want to use MariaDB and/or RRD (RoundRobin DB) to keep my data long term (https://en.wikipedia.org/wiki/RRDtool)

j-kaltes, 2025-06-28:
The webserver is explained at https://www.juggluco.nl/Juggluco/webserver.html
You can turn it on in Juggluco with Left menu->Settings->Exchange data->[Web server](https://www.juggluco.nl/Jugglucohelp/Nightscouthelp.html)
Then you can receive a list of the stream glucose values with: http://127.0.0.1:17580/x/stream
Insuline, carbohydrate etc with: http://127.0.0.1:17580/x/amounts

Remendado, 2025-06-29:
Oh yeah, as they say, poverty is the mother of invention. In the end, I set up a Nightscout server, ran the Nightscout export in Juggluco, ran the python-nightscout app on the server, and yes, configured it to send data to my mqtt broker. So many unnecessary steps, but I got result and now I can see CGM data in telegram bot.

Remendado, 2025-07-03:
> If I would like to publish juggluco data with MQTT, I would setup a juggluco server and configure a mirror connection. Then a simple cronjob every minute with a command like
> 
The main purpose is to receive data from multiple devices and send notifications, analytics and statistics to messengers, DIY devices and applications. I am not sure if Juggluco server is suitable for this.