j-kaltes/Juggluco #130: NS uploader with "test V3" enabled fails with FileNotFoundEx on /api/v3/entries API call

Jackenmen, 2024-02-25:
After enabling "test V3", I get the following error message:
```
upload
java.io.FileNotFoundException: <Nightscout URL here>/api/v3/entries
```

Java documentation seems to suggest that this exception should be specific to 404 but based on my server logs, Juggluco is actually running into a 401 here which would definitely make sense with the current upload method:
https://github.com/j-kaltes/Juggluco/blob/89df3564c2b58bd2533b4d290fcff8a0e9ea56b2/Common/src/main/java/tk/glucodata/NightPost.java#L146

V3 API no longer accepts the `api-secret` header and requires the developer to request a JWT token instead and use that for authorization. A JWT token can be requested by calling the following endpoint:
```
GET <Nightscout URL here>/api/v2/authorization/request/<ACCESS TOKEN>
```
and getting the value from the `token` key. Note that this token has an expiry date that is given in the `exp` key of the response so you can only cache it for a set amount of time.

After you have a JWT token, you can use it and make the request to V3 API with the `Authorization: Bearer <JWT TOKEN>` header.

The process of authentication and authorization is also described here:
https://github.com/nightscout/cgm-remote-monitor/blob/master/lib/api3/doc/security.md#authentication-and-authorization

j-kaltes, 2024-02-25:
You turned out the be right. It has to be the case that I only tested it with `export API3_SECURITY_ENABLE=false`.
The following update works for me even with API3_SECURITY_ENABLE=true:
https://drive.google.com/open?id=1gjqc486e5pncljbo2uiBRFeKFUA8YAki
I used the v2 authorization interface already in the Nightscout server in Juggluco itself.

Jackenmen, 2024-02-25:
Yeah, the Juggluco 7.5.0-V3 seems to work properly for me, thanks for your help :)

Btw, thank you for the "Uploads less glucose values at once to Nightscout" change from 7.2.1, I used to not be able to use Juggluco with Nightscout because my server was running out of memory every few uploads without that change :smile:

j-kaltes, 2024-02-25:
Solved.

closed 2024-02-25 by j-kaltes