j-kaltes/Juggluco #131: sudden loss of broadcasting to xdrip or AAPS

under-control, 2024-03-05:
Today all of the sudden without any changes in the apps, Juggluco stopped sending results to xDrip (or xDrip stopped receiving them), Juggluco connects nicely to the sensor and shows the results, installing newest version of Juggluco (7.6.0) did not help, turning on sending directly to AAPS did not help either. 

Now looks like data is being sent with 30mins delay with 30mins lag

Juggluco version: 7.6.0 (7.0.2 before also)
Phone: Samsung s21 fe.
Android: 13 (last updated 1 Jan 2024)

[image]

[image]
[image]

j-kaltes, 2024-03-05:
What are the data source settings in xDrip and AAPS?
What is shown when you press "_Patched Libre Broadcast_" and "_Nadawanie xDrip_"?

under-control, 2024-03-06:
nothing changed there:
xDrip looks for `Libre (patched App)` and then AAPS looks for xDrip BG
below settings from juggluco
[image]
[image]

now it started to work back again during the night when I was asleep (phone chargning during night), both patched libre broadcasting for libre and xDrip broadcasting work together, as AAPS have both signals 1min from Juggluco and 5min from xDrip so turned off xDrip broadcasting, 

it stopped working in the morning, when I started using the phone. Juggluco reads signal from sensor all the time anyways.

[image]

I suspect problem with the phone itself, it started to get hot, battery drainage and maybe android cuts some of functionalities to prevent overheating

But even when I cool down the phone it does not start broadcasting, will observe that further

j-kaltes, 2024-03-06:
When Android stopped Juggluco, Juggluco wouldn't receive glucose values from the sensor. I think that something is wrong at the receiving end.

j-kaltes, 2024-03-06:
To see if Juggluco is sending the xDrip broadcast, you can run https://www.juggluco.nl/Juggluco/xDripBroadcastView.apk
and send the xDrip broadcast to that app and view the results through logcat.

under-control, 2024-03-06:
just to share current state, juggluco collect values, even shares it to watch, but broadcast with a lag
[image]
[image]

installed xDropBroadcastView.apk on phone and will try to configure adb on pc in the evening and share some logs

j-kaltes, 2024-03-06:
Why don't you show the settings of AAPS or xdrip? Probably there is something wrong.
It seems that you have turned on "Bluetooth History" in Juggluco and then Juggluco will use a WakeLock, this will mean that Juggluco probably will not be brought the sleep by Android between receiving a glucose value from the sensor and Broadcasting the glucose.

under-control, 2024-03-06:
please see configuration from xDrip and AAPS:
[image]

And here's log from adb:
> --------- beginning of main
03-06 19:31:36.650 11048 11048 I System.out: bundle content
03-06 19:31:36.650 11048 11048 I System.out: [com.eveningoutpost.dexdrip.Extras.BgSlope]<->[2.3333333432674406E-6]
03-06 19:31:36.650 11048 11048 I System.out: [com.eveningoutpost.dexdrip.Extras.Time]<->[1709737767262]
03-06 19:31:36.650 11048 11048 I System.out: [com.eveningoutpost.dexdrip.Extras.SensorStartedAt]<->[1709321489000]
03-06 19:31:36.651 11048 11048 I System.out: [com.eveningoutpost.dexdrip.Extras.BgSlopeName]<->[Flat]
03-06 19:31:36.651 11048 11048 I System.out: [com.eveningoutpost.dexdrip.Extras.BgEstimate]<->[138.0]
03-06 19:31:36.655 11048 11048 I System.out:  glucose=7.666666666666667(mgdL=138.0) delta=2.3333333432674406E-6 rate=0.14000000059604645 Dexcomlabel=Flat time=6 mar 2024 16:09:27
03-06 19:37:39.191 11048 11048 I System.out: bundle content
03-06 19:37:39.191 11048 11048 I System.out: [com.eveningoutpost.dexdrip.Extras.BgSlope]<->[3.166666626930237E-6]
03-06 19:37:39.191 11048 11048 I System.out: [com.eveningoutpost.dexdrip.Extras.Time]<->[1709737826534]
03-06 19:37:39.191 11048 11048 I System.out: [com.eveningoutpost.dexdrip.Extras.SensorStartedAt]<->[1709321489000]
03-06 19:37:39.191 11048 11048 I System.out: [com.eveningoutpost.dexdrip.Extras.BgSlopeName]<->[Flat]
03-06 19:37:39.191 11048 11048 I System.out: [com.eveningoutpost.dexdrip.Extras.BgEstimate]<->[140.0]
03-06 19:37:39.236 11048 11048 I System.out:  glucose=7.777777777777778(mgdL=140.0) delta=3.166666626930237E-6 rate=0.1899999976158142 Dexcomlabel=Flat time=6 mar 2024 16:10:26

so message from 19:31 show data from 16:09
and next message from 19:37 show data from 16:10

and it matches the xDrip chart:
[image]

looks like throttling of some queue?

phone is really getting hot, other funcionatiltes work, but this lag in broadcasting was the first symptom I've noticed

j-kaltes, 2024-03-06:
Can you show the logcat for a longer period?
You can use `grep glucose` to make it shorter:
```bash
adb logcat --pid THEPID|grep glucose 
```

j-kaltes, 2024-03-06:
It would also be helpful if you would run a logging version of Juggluco and send trace.log to me. As described under "Bug report" in https://www.juggluco.nl/Juggluco/download.html#bugreport

under-control, 2024-03-06:
> 03-06 20:53:39.438 32036 32036 I System.out:  glucose=9.5(mgdL=171.0) delta=1.6166667143503826E-5 rate=0.9700000286102296 Dexcomlabel=Flat time=6 mar 2024 16:35:27
03-06 20:53:43.929 32036 32036 I System.out:  glucose=9.61111111111111(mgdL=173.0) delta=1.6833333174387615E-5 rate=1.0099999904632568 Dexcomlabel=FortyFiveUp time=6 mar 2024 16:36:26
03-06 20:53:45.316 32036 32036 I System.out:  glucose=9.722222222222221(mgdL=175.0) delta=1.6166667143503826E-5 rate=0.9700000286102296 Dexcomlabel=Flat time=6 mar 2024 16:37:27
03-06 20:53:45.872 32036 32036 I System.out:  glucose=9.833333333333334(mgdL=177.0) delta=1.6166667143503826E-5 rate=0.9700000286102296 Dexcomlabel=Flat time=6 mar 2024 16:38:28
03-06 20:53:46.222 32036 32036 I System.out:  glucose=10.055555555555555(mgdL=181.0) delta=1.783333420753479E-5 rate=1.0700000524520874 Dexcomlabel=FortyFiveUp time=6 mar 2024 16:39:28
03-06 20:53:49.130 32036 32036 I System.out:  glucose=10.222222222222221(mgdL=184.0) delta=1.983333428700765E-5 rate=1.190000057220459 Dexcomlabel=FortyFiveUp time=6 mar 2024 16:40:27
03-06 20:57:46.853 32036 32036 I System.out:  glucose=10.38888888888889(mgdL=187.0) delta=2.2333333889643352E-5 rate=1.340000033378601 Dexcomlabel=FortyFiveUp time=6 mar 2024 16:41:27
03-06 20:57:49.792 32036 32036 I System.out:  glucose=10.61111111111111(mgdL=191.0) delta=2.516666650772095E-5 rate=1.5099999904632568 Dexcomlabel=FortyFiveUp time=6 mar 2024 16:42:28
03-06 20:57:55.899 32036 32036 I System.out:  glucose=10.777777777777779(mgdL=194.0) delta=2.799999912579854E-5 rate=1.6799999475479126 Dexcomlabel=FortyFiveUp time=6 mar 2024 16:43:27
03-06 20:57:56.184 32036 32036 I System.out:  glucose=11.0(mgdL=198.0) delta=3.149999976158142E-5 rate=1.8899999856948853 Dexcomlabel=FortyFiveUp time=6 mar 2024 16:44:28
03-06 20:57:56.472 32036 32036 I System.out:  glucose=11.222222222222221(mgdL=202.0) delta=3.433333237965902E-5 rate=2.059999942779541 Dexcomlabel=SingleUp time=6 mar 2024 16:45:27
03-06 20:57:56.717 32036 32036 I System.out:  glucose=11.444444444444445(mgdL=206.0) delta=3.6500000953674314E-5 rate=2.190000057220459 Dexcomlabel=SingleUp time=6 mar 2024 16:46:27
03-06 20:58:06.485 32036 32036 I System.out:  glucose=11.61111111111111(mgdL=209.0) delta=3.833333253860474E-5 rate=2.299999952316284 Dexcomlabel=SingleUp time=6 mar 2024 16:47:28
03-06 20:58:11.966 32036 32036 I System.out:  glucose=11.833333333333334(mgdL=213.0) delta=4.016666809717814E-5 rate=2.4100000858306885 Dexcomlabel=SingleUp time=6 mar 2024 16:49:28
03-06 20:58:13.230 32036 32036 I System.out:  glucose=12.0(mgdL=216.0) delta=4.133333365122477E-5 rate=2.4800000190734863 Dexcomlabel=SingleUp time=6 mar 2024 16:50:28
03-06 20:58:17.107 32036 32036 I System.out:  glucose=12.11111111111111(mgdL=218.0) delta=4.199999968210856E-5 rate=2.5199999809265137 Dexcomlabel=SingleUp time=6 mar 2024 16:51:27
03-06 21:02:20.345 32036 32036 I System.out:  glucose=12.277777777777779(mgdL=221.0) delta=4.016666809717814E-5 rate=2.4100000858306885 Dexcomlabel=SingleUp time=6 mar 2024 16:52:28
03-06 21:02:21.309 32036 32036 I System.out:  glucose=12.38888888888889(mgdL=223.0) delta=3.899999856948853E-5 rate=2.3399999141693115 Dexcomlabel=SingleUp time=6 mar 2024 16:53:27
03-06 21:02:22.415 32036 32036 I System.out:  glucose=12.777777777777779(mgdL=230.0) delta=4.2333332697550454E-5 rate=2.5399999618530273 Dexcomlabel=SingleUp time=6 mar 2024 16:54:28
03-06 21:02:22.826 32036 32036 I System.out:  glucose=13.055555555555555(mgdL=235.0) delta=4.4666667779286703E-5 rate=2.680000066757202 Dexcomlabel=SingleUp time=6 mar 2024 16:55:28
03-06 21:02:25.446 32036 32036 I System.out:  glucose=13.277777777777779(mgdL=239.0) delta=4.683333237965902E-5 rate=2.809999942779541 Dexcomlabel=SingleUp time=6 mar 2024 16:56:27
03-06 21:02:25.695 32036 32036 I System.out:  glucose=13.444444444444445(mgdL=242.0) delta=4.9333333969116214E-5 rate=2.9600000381469727 Dexcomlabel=SingleUp time=6 mar 2024 16:57:28
03-06 21:02:35.482 32036 32036 I System.out:  glucose=13.61111111111111(mgdL=245.0) delta=5.266666809717814E-5 rate=3.1600000858306885 Dexcomlabel=SingleUp time=6 mar 2024 16:58:27
03-06 21:03:59.528 32036 32036 I System.out:  glucose=13.833333333333334(mgdL=249.0) delta=5.733333428700765E-5 rate=3.440000057220459 Dexcomlabel=SingleUp time=6 mar 2024 16:59:28
03-06 21:04:10.922 32036 32036 I System.out:  glucose=14.055555555555555(mgdL=253.0) delta=5.9999998410542805E-5 rate=3.5999999046325684 Dexcomlabel=DoubleUp time=6 mar 2024 17:01:28
03-06 21:04:13.891 32036 32036 I System.out:  glucose=14.166666666666666(mgdL=255.0) delta=5.800000031789144E-5 rate=3.4800000190734863 Dexcomlabel=SingleUp time=6 mar 2024 17:02:29
03-06 21:07:52.898 32036 32036 I System.out:  glucose=14.222222222222221(mgdL=256.0) delta=5.266666809717814E-5 rate=3.1600000858306885 Dexcomlabel=SingleUp time=6 mar 2024 17:03:28
03-06 21:08:04.364 32036 32036 I System.out:  glucose=14.222222222222221(mgdL=256.0) delta=4.7499998410542806E-5 rate=2.8499999046325684 Dexcomlabel=SingleUp time=6 mar 2024 17:04:28
03-06 21:08:15.328 32036 32036 I System.out:  glucose=14.222222222222221(mgdL=256.0) delta=4.4666667779286703E-5 rate=2.680000066757202 Dexcomlabel=SingleUp time=6 mar 2024 17:05:28
03-06 21:08:18.666 32036 32036 I System.out:  glucose=14.333333333333334(mgdL=258.0) delta=4.3499998251597084E-5 rate=2.609999895095825 Dexcomlabel=SingleUp time=6 mar 2024 17:06:29
[... cut ...]
| 21:13:54.557 | Dexcomlabel=FortyFiveDown | 17:41:30 | 03:32:24,56 |
| 21:13:55.762 | Dexcomlabel=FortyFiveDown | 17:42:28 | 03:31:27,76 |
| 21:13:57.025 | Dexcomlabel=SingleDown    | 17:43:30 | 03:30:27,25 |
| 21:13:57.735 | Dexcomlabel=SingleDown    | 17:44:29 | 03:29:28,74 |
| 21:13:58.545 | Dexcomlabel=SingleDown    | 17:45:30 | 03:28:28,55 |
| 21:13:59.076 | Dexcomlabel=SingleDown    | 17:46:29 | 03:27:30,76 |
| 21:13:59.828 | Dexcomlabel=FortyFiveDown | 17:47:29 | 03:26:30,83 |

will install this debugging version

j-kaltes, 2024-03-06:
And the data in logcat are the same as given in Juggluco at the time mentioned after _time=_ ?

under-control, 2024-03-06:
yes
[image]
[image]

j-kaltes, 2024-03-06:
To see it more precisely, you can use Left middle menu->Export->Stream: https://www.juggluco.nl/Jugglucohelp/helpexport.html

under-control, 2024-03-07:
I've sent the logs from debugging/logging version of app to email.

Then I've also contacted Samsung support and they recommended cache cleaning:
https://www.youtube.com/watch?v=EYVknlK8Pwc

it initially helped, broadcasting lags were absent for like an hour but then appeared again, but shorter and will probably grow again as phone is getting hotter now

I think I will do backup and try restoring to factory settings

under-control, 2024-03-09:
did restoring to factory settings, with making backup using Samsung's [Smart Switch](https://play.google.com/store/apps/details?id=com.sec.android.easyMover&hl=pl&gl=US) beforehand and then restore same application build for everything 

and all works now, phone is not overheating, in the view of battery usage nothing much changed

I hope that it wasn't planned obsolescence, but this is what Samsung's support recomended, cache cleaning and then restoring to factory settings and it finally helped.

Thanks Jaap for your time and developing Juggluco.
Closing this issue, maybe it would somehow help anybody later.

closed 2024-03-09 by under-control